kanflo/opendps #1: CC Mode

Cabalist, 2017-03-09:
Any possibility of implementing Constant Current mode?

kanflo, 2017-04-03:
I will probably not do the implementation myself as I did not use CC mode in the stock FW but I would welcome the addition in a PR.

kanflo, 2017-04-14:
As CC has been requested and thanks to the findings in #4 might be quite easy I will try to implement it.

kanflo, 2017-04-17:
CC is underway, those eager to try it out can have a look at the [CC branch](https://github.com/kanflo/opendps/tree/cc). Comments are welcome!

A long press on the "set" key toggles between the CV and CC modes, indicated on the display. TFT inversion has been scrapped.

Remote control of the CV/CC modes are still TODO.

Cabalist, 2017-04-17:
I'm just going to take this moment to say I really appreciate your work on this.  It's been really great having these in our shop and the addition of constant current really rounds it out.

kanflo, 2017-04-17:
Thanks for the kind words, appreciated!

kanflo, 2017-04-18:
Here's the current UI for setting the current/voltage in CC mode and CV mode (obviously you cannot set the voltage in CC):

[image]     [image]

kanflo, 2017-04-18:
Remote control via UART in place:

```dpsctl.py -d /dev/cu.usbserial-A9MXPNJB --mode cv```

kanflo, 2017-04-20:
For anyone testing: some runtime adjustments are needed to set the DHR12R2 register controlling CC. On one of my DPS:es, the displayed output current matches the multimeter while the other DPS has a positive  ~100mA offset. The OpenDPS FW needs to set a low value and increase until the ADC tells us we are spot on.

rletendu, 2017-04-24:
Great work on this project ! Was hopping someone succeed to open the way to hack those modules !
May be I missed something, but so far looking to the code  and the comments for CC mode I am wondering if the CC implementation could not be improved so that it behaves like a regular DPS, meaning that moving from CV to CC could be automated in default dps control.
By default DPS starts in CV and swithes to CC if Iload > Iset. CV/CC is a dps status not a control.
Sorry if my understanding of the implementation is not correct. Awaiting for my modules to arrive and test/contribute by myself.

kanflo, 2017-04-24:
Thanks! If someone comes up with a good name we could have three modes, CV, CC and "the mode you describe" ;) It's a bit of an extension of CV that does not cut power on case of over current events, something I personally prefer when testing new PCBs. I am however open to renaming any..all modes if someone comes up with better naming.

nunojpg, 2017-04-24:
Hi :) Thanks for the software. I don't have a piece of this hardware yet but when I found it I immediately searched for a open software to replace it. I've been watching the repo for a while. Eventually I will use it.

What @rletendu means is the default way a Lab power supply works. It's what "everyone" is actually expecting for a thing like this. Or I just come from a lab world...

We can start a name discussion to give to the several modes, and I provide a initial suggestion:

- Lab(oratory) / Constant Current Limiting - Reduce voltage as required to mantain the current below the specified limit.
- Fuse - when overloaded just switch off until reseted
- Hiccupp  - switch off and restart periodically (for example every second) until fault is removed

In all modes you set a current and voltage. OVP and OCP can be used for different scenarios and actually this hardware might not be capable of doing them as they are generally expected in a lab PS.

kanflo, 2017-04-24:
Thanks for the suggestions. "Fuse" sounds a lot better than CV IMHO.

rletendu, 2017-04-25:
For me a key point is to really make the difference between the operating mode requested by the user and the operating status of the power supply.
We can also define how the dps should behave when the compliance limit is reached (error mode behavior).

By default a standard lab dps is intended to be used in Force V mode and operate in CV mode when the Iload is bellow the max limit current.
When the Iload reaches the limit a regular DPS lab swithes to CC mode. But we can also provide more error mode as suggested before (fuse: swith off or periodic restart attempt).

User may want to force voltage (ForceV mode or source V mode) :
- User set a target operating voltage and a max current limit (compliance)
- In none error mode (Iload < Max current) status is "CV"
- In error mode (Iload > MAx current):  DPS swithes to CC mode (regular dps error mode behavior ) or fuse mode or Hiccupp, depending on the selected error mode

User may want to force current (ForceI mode or source I mode) :
- User set a target operating current and a max voltage limit (compliance)
- In none error mode (V < MaxVoltage) status is "CC"
- In error mode (V > MaxVoltage): DPS swithes to CV mode (regular smu error mode behavior ) or fuse mode or Hiccupp ...

nunojpg, 2017-04-25:
I went to check how the factory firmware works. It seems it works as any other standard lab power supply. So V and A limits always apply, and PS works in CV if A<set point or CA if A=set point.

OVP and OCP provide the "Fuse" mode behaviour.

So in this case there is not much to discuss about "modes". About the UI I also agree that the LCD should say if it is working on CV or CA, and OVP and OCP limits should be displayed in small font if they are enabled.

Once again, as any other lab power supply does, and also as the factory firmware does. No reason to invent anything :)

rletendu, 2017-04-27:
Agree think we should stay with a regular lab power supply dehavior for this feature.  If one wants to implement other modes he could implements its own in another feature.

avion23, 2017-06-04:
The cc branch is currently not compilable. This is the output:
`/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: opendps.elf section '.data' will not fit in region rom`

`/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: region 'rom' overflowed by 1544 bytes
`
This is on the current cc branch at commit 24ca7a2a87baf7e8409e115594c7008b95ae7f28 from Tue Apr 18 23:58:32 2017 +0200.

I've tried to investigate where the memory is lost but not

kanflo, 2017-06-05:
That's strange, 1.5kB is quite a lot. When building I end up with 5.8kB of data+bss. Could you provide some output from ```make -C opendps V=1``` ? Additionally, could you try hacking ```../libopencm3/lib/stm32/f1/stm32f100x8.ld``` and chang the ```ram``` section to twice the size to get though linkage and then be able to find the culprit?

avion23, 2017-06-05:
Btw thanks for a the effort you put into this project. Here is the verbose make output:

> make -C opendps V=1
make: Entering directory '/home/avion23/Documents/programming/embedded/stm32/opendps/opendps/opendps'
Using ../libopencm3 path to library
  CC      ui.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o ui.o -c ui.c
  CC      hw.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o hw.o -c hw.c
  CC      pwrctl.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o pwrctl.o -c pwrctl.c
  CC      event.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o event.o -c event.c
  CC      past.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o past.o -c past.c
  CC      tick.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o tick.o -c tick.c
  CC      tft.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o tft.o -c tft.c
  CC      spi_driver.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o spi_driver.o -c spi_driver.c
  CC      ringbuf.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT=15000  -DCONFIG_DEFAULT_VOUT=5000 -DCONFIG_DEFAULT_ILIMIT=500 -DCONFIG_WIFI -DCONFIG_SERIAL_PROTOCOL -MD -Wall -Wundef -DSTM32F1 -I../libopencm3/include -I../libopencm3/include/libopencm3/stm32 -I../libopencm3/include/libopencm3/cm3  -o ringbuf.o -c ringbuf.c
  CC      ili9163c.c
arm-none-eabi-gcc -Os -std=c99 -g -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -I. -DCONFIG_DPS_MAX_CURRENT= [... cut ...]
make: Leaving directory '/home/avion23/Documents/programming/embedded/stm32/opendps/opendps/opendps'

The resulting file is huge
> ls -l  opendps/opendps.elf 
-rwxr-xr-x 1 avion23 users 680392 Jun  5 11:24 opendps/opendps.elf
compared to your precompiled file
> ls -l precompiled/opendps.elf 
-rwxr-xr-x 1 avion23 users 423771 Mar  8 21:14 precompiled/opendps.elf

I've attached the opendps folder with the *.o files etc. 
[compiled.zip](https://github.com/kanflo/opendps/files/1051642/compiled.zip)

kanflo, 2017-06-05:
Very odd indeed. Looking at the output of ```arm-none-eabi-objdump -hf``` for your ELF:

```
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         0000fc90  08000000  08000000  00010000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
```

and mine:

```
  0 .text         0000e600  08000000  08000000  00010000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
```

so the text segment of your binary is 5.6kb larger... I tried building with gcc 5.4.1 from [ARM's developer site](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) but did not observe the difference you see. Could you try a more recent version?

avion23, 2017-06-07:
Oh... I think I can do even better than this. My arm-none-eabi-gcc-7.1.0 delivers:

>opendps/opendps.elf:     file format elf32-littlearm
architecture: arm, flags 0x00000112:
EXEC_P, HAS_SYMS, D_PAGED
start address 0x08003905`
>
>Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00015114  08000000  08000000  
[compiled-7.1.0.zip](https://github.com/kanflo/opendps/files/1058117/compiled-7.1.0.zip)

00010000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, CODE

With a fresh git clone on the main branch. The only change is 64K -> 128K in the ld file and removal of stm32-f7 target in libopencm3 as the assembler doesn't recognize this CPU.

Cabalist, 2017-06-28:
I noticed that the branch for this feature hasn't seen an update in a bit.  Is it currently functional?  I see there were a couple other issues open that seem daunting for this device...

kanflo, 2017-06-28:
@avion23 it seems strange you need to clean the libopencm3 build. Could you attach a small part of the output from ```make V=1´´´?

kanflo, 2017-06-28:
@Cabalist I haven't found the time to work on CC since end of April and have a few other things I need to finish before resuming it.

kanflo, 2017-06-28:
@avion23 It seems strange you need to clean the libopencm3 build as there should be anything else than F100 stuff included in the OpenDPS build. Could you please provide the output from the failing assembler?

avion23, 2017-06-29:
Ah Sorry, I managed to compile it with the official ARM toolchain. No problem there. Crossdev and gcc-arm-none-eabi from debian stretch both fail.

kanflo, 2017-07-05:
@avion23 glad you managed to sort it out.

geekbozu, 2017-10-19:
What is the current status of this? I see there have been no updates on this for a while, and might be willing to put some work in on it.

kanflo, 2017-10-23:
@geekbozu I have been quite restricted on time and have not worked on this for a while. My next step would be to merge the CC branch to master as they have diverged a bit.

geekbozu, 2017-10-27:
Do you have a rough eta on when that would be? I would hate to work on new features and make the merge more complicated then it already seems to be.

kanflo, 2017-11-09:
I am rather hoping to merge CC to master this month. We'll see...

clickworkorange, 2017-12-11:
> About the UI I also agree that the LCD should say if it is working on CV or CA, and OVP and OCP limits should be displayed in small font if they are enabled.

How about just changing the colour of the value that is being "constrained"? E.g. if current draw exceeds the set current - thus triggering CC mode - just change the current readout to yellow.

snhstq, 2018-01-15:
First : Great project, love it ! 

I want to use a 5015 with cc to charge my calcium batteries.

When I compile master (on Raspberry Pi using arm-none-eabi) it will fit and work, but I get this message when compiling the cc tree:

_****~/opendps# make -C opendps flash
make: Entering directory '/root/opendps/opendps'
  LD      opendps.elf
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: opendps.elf section `.data' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/5.4.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by **2288** bytes
collect2: error: ld returned 1 exit status
../libopencm3.rules.mk:198: recipe for target 'opendps.elf' failed
make: *** [opendps.elf] Error 1
make: Leaving directory '/root/opendps/opendps'**_**

Any idea where I might 'find' 2288 bytes ? 
I can remove the CV/CC images (2x480 bytes), but I am then still ~1200 bytes short.
Last question: where can I find a pre-complied cc for the 5015 ?  

Thanks Henk

kallemooo, 2018-01-15:
Hi @snhstq ,

The problem is that the Debian package gcc-arm-none-eabi version 15:5.4.1+svn241155-1 that provides GCC cross compiler for ARM Cortex-A/R/M processors is to old and is not usable. (gcc version 5.4.1)

Install the official toolchain from ARM and use that instead. (gcc version 7.2.1)
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

Also se issue #22 for more info.

geekbozu, 2018-02-27:
any progress here?

kanflo, 2018-04-09:
I am currently porting CC to master. Since the branches have diverged quite a bit it will not me a merge per-se.

kanflo, 2018-04-13:
First chunk of CC code is now on master. I have only tested it on one unit and the CC is a bit off, calibration might be needed.

kanflo, 2018-10-01:
Closing this one as CC is now on master.

closed 2018-10-01 by kanflo