kanflo/opendps #4: Issues with driving higher loads

kanflo, 2017-04-07:
When connecting a 4.7ohm 10W resistor at 5V, voltage drops to 1.4V. Currently the OpenDPS firmware is incapable of driving higher loads. Why remains to be found out.

kanflo, 2017-04-10:
With the PSU I have available I can drive 2A at constant voltage. Above 2A the measly PSU chokes and its output voltage drops. This issue will be left open until I am confident OpenDPS performs as good as the stock FW at higher loads.

Cabalist, 2017-04-10:
I have a flashed and an unflashed DPS5005 as well as two identical power supplies rated for 48V/9.6A. (Dell EPS-470)  

Anything I can do on the testing side to help?

Spudmn, 2017-04-11:
You are on the right path with your fix. One DAC controls the output voltage and the other DAC controls the current limit. The patch has increased the current limit to what looks like 2A.  When the current limit is exceeded the output voltage will drop.

joh, 2017-04-11:
The TL594 which controls the PWM duty cycle contains two error amplifiers. The two amplifiers are ORed so that the minimum pulse width dominates. Typically one is used to regulate output voltage while the other regulates output current.

Section 5.2 of this app note shows a typical circuit: http://www.ti.com/lit/an/slva001e/slva001e.pdf

kanflo, 2017-04-11:
@Spudmn: that would explain why I cannot get more than 2A. On the other hand, DAC2_OUT is saturated at 0xfff.

@Cabalist: I got my hands on an unflashed '5005 and will investigate how the stock FW sets ADC1/ADC2 to make the DPS output > 2A. If you have soldered JTAG to the unflashed one feel free to try the same.

@joh: thanks for the app note.

kanflo, 2017-04-12:
Tonight I checked a stock DPS5005 with the following results:

* At 4.40V DOR1 is 1 when OFF and ~0x140 when ON (0x140 pretty much consistent with OpenDPS).
* Regardless of the tested current draw (20mA, 0.9A, 1.75A and 2.56A), DOR2 was 0x73 when OFF and 0xda2 when ON.
* DOR2 remained at 0xda2 when I opened the terminals which leads me to believe there is no regulator algoritm running.

As my PSU cannot deliver more than ~2.5A @ 4.4V I will borrow a more powerful one from the lab at work to test further.

joh, 2017-04-12:
Interesting, what about with constant-current mode?

joh, 2017-04-13:
Just tested the fix on master and I'm able to draw 5A through a 1 ohm resistor without problem. I had to build with MAX_CURRENT=6000 to prevent the initial inrush current to trigger the over current condition. So I think this issue can be closed :)

Would still be interesting to learn more about how the stock firmware controls DOR1/DOR2 for constant-current mode though, but I guess that discussion can continue in #1.

kanflo, 2017-04-14:
Well that was great news, thanks! I am thinking CC could be simple to implement and will experiment with the stock FW in CC mode.

closed 2017-04-14 by kanflo