kanflo/opendps #6: Handle inrush current wrt. OCP

kanflo, 2017-04-18:
For resistors drawing, say, 400mA, the current limit needs to be set to 600mA or the OCP will kick in immediately when enabling power.

The inrush current needs to be handled, eg. by accepting inrush current > I_limit for a short period of time.

vogelfreiheit, 2018-01-14:
Does this refer to the initial spike when turning the output on? I observed a spike of 4V with output set to 5.20v initially which could already damage/toast some sensitive components.

kanflo, 2018-05-05:
Yes, I have seen the spikes too. Someday I will look into this, perhaps a ramp of a few ms could solve this.

Xenoamor, 2018-12-29:
I was thinking of this problem just recently. Perhaps having a fixed period of ~30ms and then in the settings you can adjust the percentage overcurrent it will allow for that period