kanflo/opendps #7: Implement workaround for voltage drop at high currents

kanflo, 2017-04-20:
There is an issue with the DPS5005s (at least the three I tested so far) resulting in output voltage drops at high currents. If outputting 6V (according to UI settings and UI feedback) and drawing 1.2A, a multimeter will show a 340mV drop on the output voltage compared to the UI. The value on DAC1 needs to be increased to maintain 6A @ 1.2A which also increases the displayed voltage ADC measurement.

More measurements [here](https://docs.google.com/spreadsheets/d/1AhGsU_gvZjqZyr2ZYrnkz6BeUqMquzh9UNYoTqy_Zp4/edit?usp=sharing) under the 'Voltage drop at high current draw' tab

The following table shows DAC1 values maintaining an output voltage of 6V under different current draw:
```
I [mA]   DAC1
-------------
   16    430
  133    432
  416    436
 1263    456
 2494    468
```

This is not OpenDPS related as the stock firmware has the same "feature". Fixing it would be a nice OpenDPS addition.

lumostor, 2017-08-31:
Might be due to the voltage drop in the shunt resistor, for Iout sense, which is supposed to be low side (between PSU ground and Vout-). Which mean the ADC1_in9  is reading Vout + Vshunt, because the ADC sense a value relative to the PSU ground and not Vout-. Can't test this at this moment, I have no dpsxxxx at hand but I have 3 on the way. 

According to the voltage drop of 340mV@1.2A, Rshunt should be 0.28ohms.

A fix would be to substract Vshunt to the value read by ADC1_in9, and this will be Vout.

[image]

lumostor, 2017-08-31:
I forget about the DAC, a fix would be to add the current Vshunt to the requested Vout, hopefully this would be stable, but this may need more thoughts, and surely a verification of the transients when the charge change rapidly. This might be tricky.

kanflo, 2017-10-06:
Thanks for the description @lumostor. For some reason I reraly get Github notification emails on my own repos, even though I receive tons of them each day for other repose I watch.

Xenoamor, 2019-01-02:
@lumostor did you get a chance to look into this? I think as we know the current through the shunt resistor we should be able to account for this quite easily. It does require us to know what that resistance is of course