kanflo/opendps #10: DPS5015 display freezes

toglader, 2017-07-03:
My DPS5015 display freezes every time if I press buttons too fast. Firmware self still seems to work because I can switch output on and off but display doesn't update.
I tried to debug it my self but this hw platform is new for me, so I haven't managed to setup debugger yet..

avion23, 2017-07-03:
My DPS5015 does the same. I thought it was my bad compiler and changes in the CC mode, see issue #1.

While the display is stuck at the previous value, the buttons still work

kanflo, 2017-07-05:
That's odd. Do any of you have a UART connected for debug output? If so, I'll have a look and see if there are any printf:s that can be added to debug this.

avion23, 2017-07-05:
Yes, I can test this on the weekend.

vahegan, 2017-07-14:
Seems I am witnessing a similar problem, also on DPS5015
The output enable button is not very responsive, I need to press it slowly and sometimes it works (on release), sometimes there is no effect after pressing, and a second or third press does the job. If I press that button too fast several times, the output indicator in the bottom right corner locks in whatever state it was. The output is still toggled but the indicator stays either on or off all the time. If I continue pressing the button several times (10-20 times), at some point the screen goes white and I have to cycle power to restore the functionality.

kanflo, 2017-07-14:
Hi @vahegan. Could you try building with ```make COMMANDLINE=1```which removes the binary protocol from the UART and add some printfs in eg. ```event_put(...)``` and ```event_get(...)``` so we can see what is going on with the button presses? I tried hammering the enable button on a '5005 and see no problems.
 
@gojimmypi: Have you seen this issue too?

vahegan, 2017-07-14:
Hi Johan. I can recompile but I am absolutely new to this platform and don't know which files to modify and what printfs to put and where. If you can send me a modified file, I can do that on Monday and will post the UART output.

cleverfox, 2017-07-16:
I have same issue. Before i found this thread I found this: Controls are working when LCD in freeze. I tried add command to reinit LCD into CLI (ili9163c_init(); and set roataion), and after reinit LCD continue to work. Also after reading this topic I checked event_put and event_get functions. Nothing changed between working and freezed states. I will dig bit more.

kanflo, 2017-07-19:
@cleverfox, could you see if the LCD is connected in a different manner? I recall the 5005 LCD had its SPI nCS pin grounded and I am thinking this might have changed.

cleverfox, 2017-07-19:
I desoldered LCD but there is no pin labels on it :(.
I found 8 pin header on LCD and and 8 pin cable between blocks.
here is pinout
LCD pin / cable pin / MCU pin
1 / 1 / GND
5 / 2 / pin B7 (43) inverted by transistor 
7 / 3 / VCC
3 / 4 / B12-NSS (25)
4 / 5 / B13-SCK (26)
8 / 6 / B14-MISO (27)
6 / 7 / B15-MOSI (28)
2 / 8 / A8 (29)

cleverfox, 2017-07-19:
ooops, i misaligned screen in this pinout. 1 -> 8, 2->7, 3->6 etc

cleverfox, 2017-07-19:
@kanflo you was right about slave select pin. I have added handling of NSS pin (A8), and I can't reproduce LCD hanging anymore. Beta testera are welcome to test it at cc branch :)

cleverfox, 2017-07-19:
Interesting thing: When I was testing a hypothesis about NSS, I added just gpio_set(...); gpio_reset(...); before SPI transfer and it also solved problem. But of cource before commit I made NSS handling as it should be.

vahegan, 2017-07-19:
Thanks, @cleverfox , I will test it tomorrow morning.

vahegan, 2017-07-20:
@cleverfox can you help me? I am unable to compile: in graphics.h there is an include of cl.h, and cl.h is not available

cleverfox, 2017-07-20:
Oh, yeah. I don't done image yet. You can add temporary hacked image. It is a little bit corrupted CV to distinguish mode from others. Can anybody help me with picture with CL letters like CC and CV at bottom of screen?

cl.h

`
const uint8_t cl[] = {
  0x00, 0x0f, 0x00, 0x00, 0x00, 0xff, 0x18, 0xff, 0xff, 0xa6, 0x18, 0xe3, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x0f, 0x00, 0x00, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x0f, 0x21, 0x04, 0xc6, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xbd, 0xf7, 0x84, 0x30,
  0xff, 0xff, 0x21, 0x04, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x21, 0x04, 0xff, 0xff, 0x7b, 0xcf,
  0x00, 0x00, 0xde, 0xfb, 0xf7, 0xff, 0x63, 0xff, 0xff, 0x65, 0x52, 0x8a, 0x73, 0xce, 0x52, 0xaa,
  0xff, 0xff, 0x4a, 0x69, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x4a, 0x69, 0xff, 0xff, 0x42, 0x28,
  0x52, 0x8a, 0xff, 0xff, 0x63, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x21, 0x04,
  0xff, 0xff, 0x7b, 0xcf, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x7b, 0xcf, 0xff, 0xff, 0x10, 0x82,
  0xad, 0x55, 0xf7, 0xde, 0x10, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0xe7, 0x5c, 0xad, 0x75, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0xad, 0x75, 0xde, 0xfb, 0x00, 0x00,
  0xce, 0x59, 0xd6, 0xba, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0xad, 0x95, 0xdf, 0x1b, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0xdf, 0x1b, 0xa5, 0x34, 0x00, 0x00,
  0xe7, 0x3c, 0xc6, 0x38, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x73, 0xae, 0xff, 0xff, 0x18, 0xff, 0x00, 0xff, 0xff, 0xc3, 0xff, 0xff, 0x63, 0x2c, 0x00, 0x00,
  0xf7, 0xde, 0xb5, 0xd6, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x31, 0xa6, 0xff, 0xff, 0x52, 0xff, 0x00, 0xff, 0xff, 0xaa, 0xff, 0xff, 0x29, 0x45, 0x00, 0x00,
  0xe7, 0x3c, 0xc6, 0x58, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x20, 0xef, 0x9d, 0x94, 0xff, 0x00, 0xff, 0xff, 0xb2, 0xe7, 0x3c, 0x00, 0x00, 0x00, 0x00,
  0xce, 0x79, 0xde, 0xfb, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0xad, 0x75, 0xce, 0xff, 0x00, 0xff, 0xff, 0x99, 0x9d, 0x13, 0x00, 0x00, 0x00, 0x00,
  0xa5, 0x54, 0xff, 0xff, 0x18, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x6b, 0x4d, 0xff, 0xff, 0x29, 0xff, 0xff, 0xff, 0x5a, 0xeb, 0x00, 0x00, 0x00, 0x00,
  0x4a, 0x49, 0xff, 0xff, 0x7c, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x21, 0x24, 0xff, 0xff, 0xb5, 0xff, 0xff, 0xde, 0x10, 0xa2, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0xce, 0x99, 0xf7, 0xff, 0x7b, 0xff, 0xff, 0xa6, 0x42, 0x48, 0x8c, 0x71, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0xd6, 0xff, 0xff, 0xff, 0xff, 0x58, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x10, 0x82, 0xb5, 0xff, 0xf7, 0xff, 0xff, 0xff, 0xff, 0xff, 0xce, 0x79, 0x08, 0x41,
  0x00, 0x00, 0x00, 0x00, 0x8c, 0xff, 0xff, 0xff, 0xff, 0xef, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0x08, 0xff, 0xff, 0xc6, 0x18, 0xe3, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0x00, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

uint32_t cl_len = 480;

uint32_t cl_width = 16;

uint32_t cl_height = 15;

`

vahegan, 2017-07-20:
@cleverfox thanx! Seems to work without problems now. And special thanks for bringing the calibration constants into the header file, I was looking into something like that. Do you have any suggestions about the calibration procedure? I have about (50mV + 1% of the output voltage) difference from the value sent to the DAC and the actual output voltage and that makes me a bit nervous. :)

cleverfox, 2017-07-20:
I have +/- 0.1v mistake too. But I did not touched voltage calibration yet. I will do it soon. I just made current calibration, because DPS5005 parameters is absolutely uncompatible with dps5015 (it displays value under 0.5A when real current already 4-6A)

vahegan, 2017-07-20:
How do you calibrate for current? Current seems to be not working for me at all. When I short circuit the output in CC mode and then press output enable, there is no current flowing. When I first press the output enable and then short the output terminals, there is a spark (the capacitors discharge), the screen flashes white and the protection disables the output. I haven't figured it out yet how to switch it to CL mode; in CV mode it disables the output when I short the output terminals.

cleverfox, 2017-07-20:
I don't know why it not working on short circuit yet, I seen this problem, but not investigated it yet. I tested it on few meters of LED strip. For let strip I tried current under 6Amps and it was nearly same as showing my ammeter.
Try to use any load instead of short circuit yet, but I will try to fix it soon.

cleverfox, 2017-07-20:
@vahegan I fixed overcurrent protection for short circuit in CC and CL modes. I disable protection for 2ms after start (as for now for all modes).
To all: Let's decide if we need to do 2ms delay only for CC/CL mode or it is ok for all modes?

cleverfox, 2017-07-20:
@vahegan I added example of calibration calculations inside dps-model.c.

vahegan, 2017-07-21:
Thanks, @cleverfox 
Now I see a very strange behavior. I set it up for 2.1A current output at about 10V. When I power enable with short circuit, it works as expected. But when I apply a load (a car light bulb) the screen flashes and protection is activated. However, if I short the light bulb terminals, then power enable, then remove the short from the terminals, the bulb goes on.

cleverfox, 2017-07-21:
Is this problem CL mode? I can check it only today at evening. Have you a serial terminal? I can give you patch for debug it?
Did you checked voltage after my calibration?

vahegan, 2017-07-21:
I see this both in CC and CV/CL modes. I soldered the serial wires to the board but have not connected it to the computer yet. I have a Serial2USB adapter lying somewhere, need to find it, but I will be able to do it on Monday. 
I have not done the calibration yet, only compiled it and was testing the protection. Will play with calibration later today and will let you know, thanks!

vahegan, 2017-07-21:
I just remembered another thing I wanted to mention: 
In CC mode it would be useful instead of -.-- on the voltage line to display the actual voltage measured on the load. 
Also, in setting mode for CC it would be great to have the possibility of setting the max voltage limitation i.e. the maximum voltage after which it would switch to CV mode if the current setting cannot be achieved (similar to switching to CC in CL mode).

cleverfox, 2017-07-21:
@vahegan in case if I add voltage setting in CC mode it will become same as CL mode :) I personally not fully understand what for CC mode is really useful. But I not going to destroy something already done by someone other and I created new mode CL.

vahegan, 2017-07-21:
I seem to have fried my board. I pressed output enable several time and got a white screen (just like before mods by @cleverfox ), but the board was working. I then cycled power and again pressed power enable several times. Then the screen went black and the board died. When I cycle power even LCD backlight is not available anymore and when I try to launch openocd I get an error "Inti mode failed (unable to connect to the target)". Don't know what caused this. Will try to check if there is power on the chip legs - maybe the board's power supply died.

cleverfox, 2017-07-21:
I'm so sorry this happened, but I don't think the cause was the firmware. Check voltage on C27 capacitor (near LCD socket), it should be 3.3v.

vahegan, 2017-07-21:
Yes, seems it was a coincidence, has nothing to do with the firmware. I think XL7005 is dead: there is voltage on its input but no 1.25V on the feedback pin, which means there is no switching. Unfortunately this chip is only available in China, which means 2-3 weeks wait time. I have another DPS5015 board but I would like to keep it with factory firmware until OpenDPS is fully functional.

kanflo, 2017-07-21:
Glad to have the display freeze sorted out, I will close this issue. Sorry to hear about your '5015 @vahegan.

closed 2017-07-21 by kanflo

vahegan, 2017-07-22:
@kanflo I am afraid too early to close as I am not sure its fully sorted out: I have seen the same behavior with white screen but buttons working just before the unit died. Further testing is required. I will do if my board works from external power supply to the CPU.