kanflo/opendps #11: Compatibly with DPH3205?

felixekman, 2017-07-07:
Has anyone tried this with the DPH3205 which is the buck-boost converter version of this?

kanflo, 2017-07-19:
I have none to test. Adding support should be a breeze though as the STM32 is not covered in a sandwich PCB design as with the 5005. Really helps the reverse engineering effort.

felixekman, 2017-07-20:
Do you have any guidence on how one would go about starting this effort to be compatible with your software? I'm willing to put down the needed time to make it happen, just need some pointers on how to best contribute.

kanflo, 2017-07-21:
You can start off with [my reverse engineering effort](https://johan.kanflo.com/hacking-the-dps5005/) and learn how the stock FW controls your device. You need to figure out how the buttons are connected and how the output voltage and current are set. For the DPS5005 the DAC is used for voltage and TIM2 (IIRC) for current limit.

For the voltage setting, I would dial a selection on the stock FW and then check the DAC setting using the ocd-client.py script with OpenOCD connected. For your information, the stock FW freezes from time to time when JTAG halted.

cleverfox, 2017-10-08:
@felixekman Did you tried it?

gnbl, 2018-12-04:
Info on DPH series:

https://lygte-info.dk/review/Power%20DPH5005-USB%20UK.html
https://lygte-info.dk/review/Power%20DPH3205%20UK.html

And my own register/memory dumps (made with https://github.com/kanflo/opendps/blob/master/ocd-client.py) for STM32F100C8T6B on DPH5005, hardware V1.1 with firmware V1.6:
[DPH5005_HW=V1.1_FW=V1.6.zip](https://github.com/kanflo/opendps/files/2643866/DPH5005_HW.V1.1_FW.V1.6.zip)

Grounding PA6 causes fan to start, so this appears to have similar temperature sensing as DPS5015 (see info in https://github.com/kanflo/opendps/issues/50).

opeRaptor, 2019-02-25:
Are you working on a port @gnbl ? I am interested in this version as it is buck-boost.

gnbl, 2019-02-25:
@opeRaptor Not since my last post and not again any time soon, so go ahead.

polihedron, 2019-10-25:
#197

Harvie, 2019-10-25:
Are the differences between DPS and DPH so big? If someone convinces me that he/she is capable of porting opendps to DPH i will send him DPHxxx device for free.

Wegerich, 2023-10-03:
We have a DPH3205 for our community repair group. It's used by many repair volunteers so it would be amazing to have a better interface where changing current and voltage is more intuitive.