kanflo/opendps #13: SCPI support needed?

cleverfox, 2017-07-20:
Hello guys. I would like to ask your opinion about SCPI support for opendps. 

SCPI (Standard Commands for Programmable Instruments) is industrial standard for controlling of measuring equipment and also programmable power supply over Serial/USB/GPIB/Ethernet. As for opendps I can implement serial version of SCPI and it will be compatible with true serial (RS-232), USB (over USB-UART adapter) and even ethernet (wifi over esp8266 or RTL8710AF).

This give us ability to use standard software (anybody using it already?). As for me, I using whis protocol with my oscilloscope, but my software have no support for anything apart this oscilloscope.

You can see example command set for power supply here http://www.manson.com.hk/getimage3/index/action/images/name/SDP_SCPI_command_list_1_2.pdf 
and https://github.com/eez-open/psu-firmware/blob/master/doc/SCPI%20reference%20guide/EEZ%20PSU%20SCPI%20reference%20guide.pdf

kanflo, 2017-07-21:
I think SCPI would be very useful. The eez-open project is really cool and puts this side project of mine to shame ;)

Robin10155, 2018-06-17:
Hi cleverfox,
      Are you working on this new feature? I think it would a great addition to opendps.

cleverfox, 2018-06-30:
No, I have no time right now. I will have a time at the end of this year. If you like you can take this task :)

kanflo, 2018-06-30:
For anyone wanting to add SCPI support, I can recommend [this library](https://github.com/j123b567/scpi-parser) by @j123b567.

Spudmn, 2018-07-04:
I am going to add SCPI support next.  I will let you know when it is ready

j123b567, 2018-07-08:
@Spudmn I know that this is just early version so take this only as a hint.

Beauty of SCPI is that it is standard and you can relay on some conventions. For example, if I work with power supply, I expect, that it implements "Power Supplies" instrument class. I also expect that SCPI instrument implements all mandatory commands from IEEE 488.2 and SCPI-99 even if they are just stubs.

If you don't like this, you probably don't need SCPI and this SCPI library is overkill for you. By the way, OpenDPS already has simple custom nonstandard remote commands so why to create another complicated nonstandard remote commands set?

If you like SCPI (I hope so), please implement ["Power Supplies" instrument class from the specification (SCPI-99 - Volume 4: Chapter 7 - page 792)](http://www.ivifoundation.org/docs/scpi-99.pdf) and keep all mandatory commands even if you currently don't understand, why they exists. Using default implementations of mandatory commands is enought for the start.

Spudmn, 2018-07-09:
@j123b567  Thanks for the hint. 
Yes I like SCPI and I intended to implement the mandatory commands.  Thanks for the PDF.  I am sure that will be very useful.

kanflo, 2018-07-10:
Cool, looking forward to see the complete command set. Please try to follow the coding style so far. That is would be

```
static scpi_result_t dps_output_enable(scpi_t * context)
{
...
  if (opendps_enable_output(param1)) {

```

rather than 

```
static scpi_result_t DPS_Output_Enable(scpi_t * context) {
...
  if (opendps_enable_output(param1))
  {

```

Xenoamor, 2018-12-15:
Sounds interesting... what computer software uses SCPI? I'd be interested to know if it's worth the effort

kanflo, 2018-12-15:
When working with test equipment SCPI is quite common. All instruments from the big guys such as National Instruments speak it.