kanflo/opendps #14: curent limit no presant , will be added ?

finos2, 2017-08-25:
open dps is undergrade without the current limiter for usage in a lab , also ad in the front panel watts mesurment

cleverfox, 2017-09-27:
Look at cc branch. And also you can read this pr https://github.com/kanflo/opendps/pull/12

kanflo, 2017-10-06:
I currently have no plans to add the watt measurement to the front panel but pull requests are always welcome ;) The cc branch will/should be merged to master at some point.

cleverfox, 2017-10-06:
@kanflo what do you think about reducing the font size? If we can reduce it to add one more line of text (probably small font), we can make opendps more flexible. 
I have few ideas about third display line: wattmeter, energy meter (useful for battery charging), and also in wattmeter mode probably useful feature is power limit (for example when checking LEDs when nominal current and/or voltage is unknown)

kanflo, 2017-10-17:
Reducing the font size is doable, I created [font-convert.py](https://github.com/kanflo/opendps/blob/master/opendps/font-convert.py) just for that purpose. I guess at some point the UI will be configurable via JSON.

roger-, 2017-11-05:
Can a voltage limit be set in CC mode?

cleverfox, 2017-11-05:
In СС branch you can set maximum voltage and maximum current.

closed 2018-03-22 by kanflo

roger-, 2018-03-22:
Will the CC branch be merged soon?

kanflo, 2018-03-22:
I am hoping so, yes.