kanflo/opendps #15: Disable button doesn't work in my DPS3003 with CC firmware

splasma, 2017-09-27:
Hi, first at all, great work !!
After burned the CC firmware in my "small" DPS3003 the enable/disable button does not  disconnect the output, the voltage is  getting out always :-( ...

kanflo, 2017-10-06:
Thanks! I recall this was a problem with the newer comms versions of the DPS5005, but got fixed in [this](https://github.com/kanflo/opendps/commit/5d2e74b90c973e079a509b304c4f636599f78d8f) commit. You did not by any chance run [ocd-client.py](https://github.com/kanflo/opendps/blob/master/ocd-client.py) before flashing as described under ["Please note" here](https://johan.kanflo.com/upgrading-your-dps5005/)? Chances are the hardware changed meaning the disabling sequence needs trimming.

splasma, 2017-10-06:
Uhh, I didn't run ocd-client.py before flashing... as you recommended (!),
next time :-)
Anyway, with the "master branch" firmware ( no CC) works perfectly.
thanks for the quick reply.

kanflo, 2017-10-17:
I guess I should modify ```make flash``` to ask if ocd-client.py has been run :)

closed 2017-10-17 by kanflo

splasma, 2017-10-17:
:-) good idea ! for impatient people...