kanflo/opendps #17: Setting voltage, Turning power on and getting a different voltage then I set?

geekbozu, 2017-10-19:
Not sure if this is an actual bug. But I got my DPS today flashed it with OpenDPS and was playing with it. I currently set the voltage to 5.00 and when I turn it on it drops down to 4.74 even with out a load attached? Is this normal/expected behavior?

kanflo, 2017-10-19:
@geekbozu That would be unexpected behavior. Is 4.74V consistent between the display and when measured with a multimeter? Did you verify the functionality of the DPS with the original firmware?

@fromeijn Could you please explain "open loop system"?

fromeijn, 2017-10-19:
With "open loop system" I mean, that the dac is set without reading it back. There is no PID controller of some sort between it.

geekbozu, 2017-10-19:
From my measurements it was accurate. I will not lie how ever I did not
verify before flashing I made sure an led lit up then flashed away. I did
save the constants for 5V-On and OFF however.If it is any consolation the
measurement of the input voltage also matches my power supply I am using.
It does seems to be in agreement with what @fromeijn is saying. As the
error is less drastic at lower voltages and gets worse as I get higher up.

kanflo, 2017-10-19:
Thanks for the explanation @fromeijn. A PID controller would be very useful, I will create a new issue to track it.

geekbozu, 2017-10-22:
So I quadruple checked. It seems that the on screen display is correct give or take .02v according to my multi-meter.  Would using a feedback look make more sense then say a per unit calibration? Would almost seem like its worth making a manual calibration function to tune each unit that way. (and to have power levels not have to adjust as much after being turned on)

kanflo, 2017-10-23:
A PID controller would be better IMHO as people have observed voltage drops at higher loads.

geekbozu, 2017-10-23:
I think I was not clear. I more meant would a way to calibrate be useful In addition to the PID controller. Or do the devices seem accurate enough from the factory that the PID tuning should be quick enough to not matter. I know I have burned components up due to power supplies that jump to higher voltages on startup before settling down. So that is where my concern is from there.

kanflo, 2017-10-24:
Hmmm, you have a point there. The controller would have to be extremely quick to react to reductions in load as not to fry things. Calibration could be a better option for those not using higher loads.

geekbozu, 2017-10-24:
Well the thought is you need both. One to handle initial startup/Hardware
discrepancies, The PID to handle variations in Load + Drift. That being
said I am more then willing to work on a calibration interface/setup if
need be once the CC branch is merged (no point in starting another project
on a split codebase IMO) and could probably tinker with the PID tuning if
@fromeijn does not jump on that.

Manawyrm, 2017-10-25:
Can confirm, with voltage set to 5.00V, it outputs 4.729V (and displays 4.71V)...
I'd also say that a calibration/offset setting would probably be better than a PID loop, which could overshoot and kill sensitive things...

fromeijn, 2017-10-25:
Good suggestion, we don't want any magic smoke. So we need a calibration routine for voltage and current.

geekbozu, 2017-10-26:
I guess I will start familiarizing my self with the code base. :D Seems like a worthwhile effort to me.

vogelfreiheit, 2018-01-14:
Did anyone get somewhere with either self-calibration or PID? I have updated calibration values for my 5015. Both DAC and ADC.

geekbozu, 2018-01-14:
I have been waiting for CC mode to be implemented to not fracture the code
more before I started working on it, I do plan on implementing it soon that
being said.

geekbozu, 2018-04-08:
So over time it seems my device has also lost its factory calibration. Im reading 7.8v when I supply it 5v.

kanflo, 2018-04-09:
For OpenDPS devices there is no factory calibration but it seems a calibration function just got pushed up the TODO list.

geekbozu, 2018-04-09:
I will dig into it a bit more once I have some time my self, and as noted I'll gladly tackle setting up some kind of calibration UI/scheme once the CC branch is "merged"

kanflo, 2018-04-13:
Adding a calibration UI scheme would be quite easy now as the UI supports different screens with editable widgets. See [func_cv.c](https://github.com/kanflo/opendps/blob/master/opendps/func_cv.c) and [func_cc.c](https://github.com/kanflo/opendps/blob/master/opendps/func_cc.c).

antonblanchard, 2018-05-27:
My DPS5005 exhibited the same issue. Before flashing I had taken dumps at a few voltage levels, so I could work out the correct V_DAC_K and V_DAC_C settings:

```
#define V_DAC_K (double)0.0776
#define V_DAC_C (double)2.0
```

My mulitmeter, while not calibrated, showed voltages were spot on after this change. I then adjusted V_ADC_K so that the voltages displayed matched:

```
#define V_ADC_K (double)12.8891
```

I tested the open circuit voltage as well as under almost 5A of load, and the voltage was consistent. This suggests the issue is just a calibration one, and a PID is not required.

vogelfreiheit, 2018-05-31:
Late to the party, I will go through the comments.... glad other people are still working on this. I'm still working towards a successful dump of the original firmware :>

Robin10155, 2018-06-26:
Can somebody please explain what does V_DAC_C and V_DAC_K means? I am trying to calibrate the DPS5005. Currently it is off by 300mV at 5V.

kanflo, 2018-06-26:
@Robin10155 It's the equation of the straight line describing the voltage output as a function of the DAC setting:

```
V_out = DAC_setting * V_DAC_K + V_DAC_C
```

See [this spreadsheet](https://docs.google.com/spreadsheets/d/1AhGsU_gvZjqZyr2ZYrnkz6BeUqMquzh9UNYoTqy_Zp4/edit?usp=sharing). You can calibrate your unit by measuring V_out given a series of DAC settings and plotting the values.

Robin10155, 2018-06-28:
Thanks for a quick reply. I will give it a slot.

kanflo, 2018-10-01:
From #48: "I see exactly the same behaviour wrt. output voltage on my new '5005. I will look into solving this at build time."

Xenoamor, 2018-10-01:
Thanks. If you find what is different on the schematic let us know!

geekbozu, 2018-10-11:
Could, At your leisure, Explain how to use the PAST library you have included as well as clarify the bug that is noted in your comments? I plan to get to this soon (finally) and want to make sure that we can save the configs to start >.>

With that would getting a command line based calibrator make more sense first? Or should we go for from panel interface-able calibration right off the bat?

geekbozu, 2018-10-13:
Yeah that is what I was thinking instead of writing on on hardware ui to start. 

Process something like such.

"Set Dac to value" -> User types in real value
x2-3 times
Once voltage is done, 
Ask user for Resistor value, -> set output voltage and measure current @ known value
x2-3 times

Save and have a well calibrated output. 

Input calibration will be a little trickier but not horrid.
Working on this now starting with getting it to load constants from past then we can extend dpsctl.

vogelfreiheit, 2018-10-14:
I second both the PID controller loop idea and also having an UI "debug menu" of sorts (text screen is plenty good) that we could activate and do the calibration there. Adding a requirement for dpsctl means another layer of complexity and extra hardware when we could literally just do this with a barebones UI that takes Vin and Vout and perhaps Cin Cout and adjust things internally.

Nobody should use the firmware AS-IS with the built-in calibration constants because there is no guarantee whatsoever of those being accurate. These values are directly tied to the hardware. Every board will differ. The original firmware uses a PID loop if I am not mistaken.

geekbozu, 2018-10-14:
I already have my local copy loading constants from flash, I'm currently working on implementing a way to send the constants via DPSCTL once that's functional I will be working on a UI wizard for it. Because I agree it needs something. 

Once we have the ability to know out constants work across the voltage and amperage range we can/will work on some kind of run time adjustment.

vogelfreiheit, 2018-10-14:
Unrelated but: we could add MQTT capabilities to the ESP32 firmware. Not sure what is available and the code quality involved, but that would make it ridiculously useful for integration with all kinds of automation systems and metrics gathering.   If some issues are ironed out, the whole package would rival (based on cost alone) pretty much anything else the average hobbyist could get his hands on.

geekbozu, 2018-10-14:
Throw it in the  SCPI support issue both would be pretty sweet

geekbozu, 2018-10-15:
Hit a small snag where ATOF is using more memory then originally intended so I need to rework how dpsctl sends commands for calibration over. But progress is still happening. Slowly in my FORK

Xenoamor, 2018-10-15:
Are there any downsides to moving from fixed point to floating point, I believe there's no floating point unit in the MCU?

Also I think there are a few lines like this: https://github.com/geekbozu/opendps/blob/c1edaaac687a703266e88f62cda0f17f033ca8cd/opendps/pwrctl.c#L275 that will lose the fractional part (i.e. 1.9 will become 1)

geekbozu, 2018-10-15:
IMO not really it will be a bit slower but not really enough to matter for what we are doing. OCP is the only thing that will be really effected by it and OCP should be set to be more forgiving anyway then @ death point of the Device Under Load. 

For the items like what was linked is not really ok but acceptable as depending on the case.  

It is a value being sent to the dac and needs to be a whole number anyway.
or
Is a value with a resolution of .01mV, The floating point is to help with rounding/accuracy down to a few uV. 

However they probably should be rounded instead of truncated so your "closer" to your real target. I'll adjust it later on if needed Not particularly relevant for this issue. Once Calibration is good we can automate some data logging to see how off it may be and if we need to make adjustments :)

Xenoamor, 2018-10-16:
I believe Kanflo pre-calculates the OCP value with pwrctl_calc_ilimit_adc() and stores it in the variable pwrctl_i_limit_raw so speed shouldn't be an issue when it comes to this

vogelfreiheit, 2018-11-04:
How is this progressing?

geekbozu, 2018-11-04:
Coming back around on the round robin soon. I need to make a "ui" still and that is really it for calibration you can manually specify the values if you know what they are with my version of DPSCTL.py (supposing I pushed the right version there was a small bug last I checked). Making a case for it then its back to making this work right so i'm not developing it on my lap.

geekbozu, 2018-11-06:
https://imgur.com/a/WwTwlj2
Welp case is made. Really happy with it so now I can sanely get back to this. Hopefully I get some free time from work soon.

Note: Line 120 Protocol.py needs to be "f" not "d" for anyone testing will be fixed in the next commit.

geekbozu, 2018-11-06:
https://imgur.com/a/WwTwlj2
Welp case is made. Really happy with it so now I can sanely get back to this. Hopefully I get some free time from work soon.
Is it possible to update the boot loader with out the SWD wire-up? I really don't want to attempt to solder that backup any time soon if I don't have to. 

Note: Line 120 in my fork Protocol.py needs to be "f" not "d" for anyone testing will be fixed in the next commit.

geekbozu, 2018-11-11:
So I have gotten Output Voltage calibration working 👍 
Input Voltage Calibration is next. 
Current/Constant current mode are after that. 

If anyone wants to test I would be grateful.
https://github.com/geekbozu/opendps/commit/46df2c9d939d98602ddfeae8ec2095003c535fc8

code for dpsctl for calibration
`dpsctl.py -d comX -c`

I added a re-init function incase your DPS gets really messed up, It reverts back to the constants supplied in dps-model.h the code for that is
`dpsctl.py -d comX --init`

Thanks!
[Testing Binary](http://img.keepdream.in/opendps.bin)

geekbozu, 2018-11-11:
@kanflo is the formula for calibrating the V_IN ADC different then V_OUT? I cant seem to produce results that match real world using the current formulas, Nor do your spread sheets match the values supplied in pwrctl.c

Xenoamor, 2018-11-12:
This is great! Thanks so much. I'll be sure to test this as soon as I'm back from an international work trip

geekbozu, 2018-11-12:
Kinda, For input voltage calibration we need to get a reference point on the ADC, We do that by feeding a known voltage to the DPS we can get known ADC values to compute our coefficients. 

I am currently still debating on how I want to handle current calibration, I might have a way to handle both CC and Current display in one shot. I'm looking over if it will work well for us.  :)

kanflo, 2018-11-12:
Really cool work @geekbozu, hoping to try this out tomorrow!

vogelfreiheit, 2018-11-14:
I'll try this over the weekend! I might need to look into soldering a ESP32 directly to the UART and have some sort of bridge or such for this purpose. Anyone has any custom firmware they recommend? It could be useful to have graphing or data output from the PSU (maybe fake SCPI?).

geekbozu, 2018-11-15:
It is not wired up in the pictures but in mine I got a USB bulkhead adapter and wired up a USB -> Serial adapter for mine. https://imgur.com/a/WwTwlj2 You can see the bulkhead in the last picture.

Xenoamor, 2018-11-19:
You might want an isolated USB to serial adaptor

geekbozu, 2018-11-19:
Only ground is wired up, So it may introduce some noise to the circuit/get some noise from the circuit but other then that it should be fine, I honestly don't plan on using it with USB plugged in in most cases anyway its really just for development and occasional data logging.

kanflo, 2018-12-14:
I finally got around to test and I am impressed! Much looking forward to your PR.

geekbozu, 2018-12-15:
I actually have been waiting for someone other then me to test this as I haven't had much time to put into it! So its been sitting on my backburner waiting for voltage calibration to be tested before I start working on current calibration!

Xenoamor, 2018-12-15:
Will test now

geekbozu, 2018-12-15:
I will gladly accept any input on how to make this program flow better. User side software is definitely not my strong point :)

Xenoamor, 2018-12-15:
Ooops... disregard that I haven't had my evening coffee. I input the values out by a factor of 10....
It seems to have corrected the 140mV offset that the device had on the input with the default values so all is working well!

I'll try the rest now

geekbozu, 2018-12-15:
Woo :) Definitely need to spruce up the instructions but for now "they work"

Xenoamor, 2018-12-15:
It seems to be working to me. I think with some code cleanup and some clearer instructions in the python tools this should be good to go. I'll start making pushes to your repo for these

geekbozu, 2018-12-15:
That would be amazing, Thank you a ton. Ill work on tackling Current calibration soonish Should have time later this week.

kanflo, 2018-12-15:
If you store the coefficients in the past they will survive power cycles. I see you store the K and C values in different units which in theory may cause problems with atomicity. I would recommend storing them in the same unit.

geekbozu, 2018-12-15:
I can add the mangling to store them into 1 past unit later. Due to the fact that there is only one place where they are written and read they should always reflect the last written value.

geekbozu, 2018-12-16:
It very much is possible to re-work that, I am not super versed on how to effectively handle blocks of code like that so I defaulted back to If/Else. Thankfully it does work in its current state and at the least should not need to be expanded to much Honestly this code needs more love IMO https://github.com/geekbozu/opendps/blob/a8d7248c63a8b69777929b5a6959c5ae0520c9af/opendps/pwrctl.c#L55

Definitely agree those routines are a mess in both the Python file and the C source code. They work but they are not really beautiful code 👍

Xenoamor, 2018-12-16:
I didn't realise that the serial commands operate on `variable=value`. Please disregard my comments on changing [opendps_set_calibration](https://github.com/geekbozu/opendps/blob/1da1c1f8eab3db899b42696d3ccb18ea233ee3d4/opendps/opendps.c#L274) to use enums. I'll look at cleaning up handle_set_calibration/handle_list_parameters

[33 more replies]