kanflo/opendps #19: Recommended header/socket for pinouts

vogelfreiheit, 2017-11-25:
I have been checking the headers and it would be very useful to advertise which JST-GH connector and socket match would fit. Both pitch and number of pins, as UART and JTAG/programmer IO are exposed together. Is it 1.25 pitch? I suppose we need both a 4P(in) and a 5P(in).

kanflo, 2017-11-25:
Yes, the pitch is correct. [JST-GH](http://www.jst-mfg.com/product/pdf/eng/eGH.pdf) is 1.25mm. On the DPS5005 I have from this fall (which _might_ have changed by now) has 5x JST-GH for SWD (only 3x are needed; GND,SWCLK and SWDIO) and 4x for comms. There is no room for soldering a male connector (I believe there is only a single one for JST-GH) for JTAG though as the one for comms partially covers the PTHs for JTAG. I have an adapter board that can be used with pogo pins for interfacing with SWD on its way from China as we speak:

[image]

vogelfreiheit, 2017-11-26:
This is pretty sweet.  I just flashed 5015 and it is not outputting voltage right now. I can upload a copy of the 5v settings on/off. Funny enough I just got a PCB order on its way... should have asked you earlier.

kanflo, 2017-11-26:
Oh, you did a similar one? :D Have you tried flashing from the CC branch? I recall that one having better '5015 support. If still nothing, feel free to open an issue.

vogelfreiheit, 2017-11-29:
Not at all, I ordered a bunch of RF and assorted misc PCBs :)

I have been checking the CC branch. There are significant changes related to GPIO handling and so on, how much of it is merged and/or adopted into master? I will flash soon, finishing some other work here. But it looks good. Is it me or the DAC code seems ready to handle constant voltage?

closed 2018-03-22 by kanflo