kanflo/opendps #20: Suggestion: leverage STM32 protected memory read attack to download fw

vogelfreiheit, 2017-11-26:
I suggest a look at this as a really fun project:

https://www.aisec.fraunhofer.de/en/FirmwareProtection.html

Long story short, we could dump the firmware. I have one remaining 50V5A while the 5015 remains non functional (I have not found the cause for the disabled power output yet).

If someone wants to send a module my way, or can do this on his own, be my guest!

kanflo, 2017-11-26:
That I must try when I buy my next 'DPS. Thanks!

perillamint, 2017-11-28:
I read paper just now. It seems it only sheds light on parts of flash memory where PC or other memory instruction travels. Also, it requires de-soldering of MCU to do this attack.

I think to execute this attack, we need some special helper circuitary to ensure stock firmware travels all needed parts of flash memory by triggering proper inputs to target MCU.

cleverfox, 2018-02-16:
There is absolutely not a problem to desolder TQFP48 and solder it to breadboard. I made this a lot of times with soldering iron and/or with hot air gun.
@vogelfreiheit what kind of problem you have with 5015? I using it often with opendps (version from cc branch before merging to master). I know about few little problems, but they don't too hurt

goosenphil, 2018-05-31:
@vogelfreiheit Have you attempted any of the attacks that you've linked to?

kanflo, 2019-02-17:
Impressive attempt never the less @gnbl. Did you make any progress?

gnbl, 2019-02-18:
No, I did not spend any more time on this.
I assume that "our" F100 is not vulnerable, as suggested by the original authors.
In addition, I think this project already did a great job with reverse-engineering, so the time that would have to be spent on disassembling and understanding the original firmware is better used for work on opendps. But of course this is an interesting subject and it would be nice to restore the original firmware if experimenting with opendps...

vogelfreiheit, 2019-02-22:
Cool you guys kept at this! @gnbl are you on IRC somewhere? I can help you out with this probably the next few days.

The useful outcome is to be able to either run the original fw after a reflash or even boot it from external NAND/EEPROM (with our own custom bootloader).

gnbl, 2019-02-22:
@v Sorry, I'm not pursuing this further.

mpictor, 2019-08-25:
FWIW, [someone on Medium](https://medium.com/@LargeCardinal/how-to-bypass-debug-disabling-and-crp-on-stm32f103-7116e7abb546) thinks the stm32f103 _is_ vulnerable...

gnbl, 2019-08-26:
@mpictor No, it's just about debug disable, and a reference to the same project regarding code protection.

gnbl, 2020-03-19:
It seems there is a different vulnerability that could be used to extract around 90% (depending on STM32F1 model) of the firmware in an hour:
https://blog.zapb.de/stm32f1-exceptional-failure/

However, at first glance it looks like the read out is fragmented (depending on STM32F1 model) and so cannot be used to restore the original functionality.

gnbl, 2020-12-07:
Obermaier et al have revisited the F1 and compatible devices (https://www.usenix.org/conference/woot20/presentation/obermaier), finding they could read 100% flash memory from all of them, including STM32F103.

https://www.usenix.org/system/files/woot20-paper-obermaier.pdf

> 7.4 H3: Shellcode Exec. via Glitch and FPB
> Affected: APM32F103, STM32F103

> For the attack, we employ solely an STM32F303 evaluation
> board and some wires. The board provides a debug interface
> for the DuT, a UART interface for serial communication with
> the shell, and it controls the Reset, BOOT, and power supply
> signals via its GPIO pins. Two GPIOs are connected in paral-
> lel to provide sufﬁcient current for the DuT. This experiment
> shows that low-cost tools are entirely sufﬁcient to circumvent
> ﬁrmware protection in the STM32F103 and APM32F103.

https://github.com/JohannesObermaier/f103-analysis/tree/master/h3
- DUT: https://github.com/JohannesObermaier/f103-analysis/blob/master/h3/rootshell/main.c 
- attack: https://github.com/JohannesObermaier/f103-analysis/blob/master/h3/attackboard/Core/Src/main.c

The "Attack board" appears to be https://www.st.com/en/evaluation-tools/stm32f3discovery.html (there are different revisions)