kanflo/opendps #22: CC branch building issues with Ubuntu 17.10

vogelfreiheit, 2018-01-09:
I have to double check but the linker is complaining. The .data section is overflowing into .rom.
I did not build the CC branch before, but this is in a new 17.10 VM. 5015 support is still broken it seems, so we will see...

kanflo, 2018-01-10:
Oh, that's weird. I will have a look at it tonight.

kallemooo, 2018-01-10:
The problem is that the Ubuntu and Debian package gcc-arm-none-eabi that provides GCC cross compiler for ARM Cortex-A/R/M processors is to old and is not usable.

Install the official toolchain from ARM and use that instead.
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

kanflo, 2018-01-11:
Thanks @kallemooo! I will close this issue, feel free to reopen it @vogelfreiheit if you run into problems.

closed 2018-01-11 by kanflo

vogelfreiheit, 2018-01-14:
Not sure if commenting will reopen this, but I will file some tickets. I checked the calibration for the 5015 model and I identified some other problems, namely the current and output voltage inconsistency. I am not sure how much was merged from the CC branch but we should sync, maybe exchange some IM coordinates and figure out a clean merge. I finally got time to do this, thankfully, and I am interested on having a more functional firmware than what we offer now.