kanflo/opendps #24: makefile can't cope with file path with spaces when flashing

cortices, 2018-02-15:
$ make -C opendps flash

Results in

```
FLASH   opendps.elf
(echo "halt; program /Users/sparky/Documents/Making Projects/DPS5015 Upgrade/opendps/opendps/opendps.elf verify reset" | nc -4 localhost 4444 2>/dev/null) || \
		openocd -f interface/stlink-v2.cfg \
		-f target/stm32f1.cfg \
		-c "program opendps.elf verify reset exit" \
		2>/dev/null
��������Open On-Chip Debugger
> halt; program /Users/sparky/Documents/Making Projects/DPS5015 Upgrade/opendps/opendps/opendps.elf verify reset
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
auto erase enabled
Invalid command argument
image.base_address option value ('Upgrade/opendps/opendps/opendps.elf') is not valid
embedded:startup.tcl:476: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 532
at file "embedded:startup.tcl", line 476
> %
```

cortices, 2018-02-15:
This is the command verbatim that did eventually work manually in the telnet session:

    program \"/Users/sparky/Documents/Making\ Projects/DPS5015\ Upgrade/opendps/opendps/opendps.elf\" verify reset

Notice \\-escaped quotes, and \\-escaped spaces. Weird.

kanflo, 2018-02-19:
Thanks, I'll look into that. Glad you found the workaround.