kanflo/opendps #25: At least some DPS5015 units have RGB displays, not BGR.

cortices, 2018-02-27:
I modified my power icon to be red, and ran the graphics convert. However, when programmed, this rendered blue.

I modified line 29 of `ili9163c_settings.h` to the following:
```c
    #define __COLORSPC		0// 1:GBR - 0:RGB
```
instead of 
```c
    #define __COLORSPC		1// 1:GBR - 0:RGB
```

and the problem was solved. This may be due further investigation into which units have which colour space, as it is not immediately apparent either way with the original black and white scheme.

kanflo, 2018-02-27:
Thanks for letting me know. I will keep this issue open until the documentation has been updated and color space can be changed when building.

kanflo, 2018-03-04:
It's now possible to add the argument COLORSPACE when building if needed:

```
% make -C opendps COLORSPACE=0
```

README will be updated shortly.

cortices, 2018-03-04:
Great. I will try to PR a merge of this into CC branch when I get the chance.

closed 2018-03-22 by kanflo