kanflo/opendps #26: Cannot clone

jandcmoore, 2018-03-09:
`$ git clone --recursive https://github.com/kanflo/opendps.git
Cloning into 'opendps'...
remote: Counting objects: 864, done.
remote: Compressing objects: 100% (26/26), done.
remote: Total 864 (delta 13), reused 19 (delta 8), pack-reused 830
Receiving objects: 100% (864/864), 1.26 MiB | 1.20 MiB/s, done.
Resolving deltas: 100% (433/433), done.
Checking connectivity... done.
Submodule 'dpsctl/uhej' (https://github.com/kanflo/uhej-python.git) registered for path 'dpsctl/uhej'
Submodule 'esp8266-proxy/esp-open-rtos' (git@github.com:SuperHouse/esp-open-rtos.git) registered for path 'esp8266-proxy/esp-open-rtos'
Submodule 'esp8266-proxy/uhej' (https://github.com/kanflo/uhej) registered for path 'esp8266-proxy/uhej'
Submodule 'libopencm3' (https://github.com/libopencm3/libopencm3.git) registered for path 'libopencm3'
Cloning into 'dpsctl/uhej'...
remote: Counting objects: 30, done.
remote: Total 30 (delta 0), reused 0 (delta 0), pack-reused 30
Unpacking objects: 100% (30/30), done.
Checking connectivity... done.
Submodule path 'dpsctl/uhej': checked out 'dbb0a8aa12c58b739a4d3e5a0a4a812187ecc6a8'
Cloning into 'esp8266-proxy/esp-open-rtos'...
The authenticity of host 'github.com (192.30.253.113)' can't be established.
RSA key fingerprint is SHA256:nThbg6kXUpJWGl7E1IGOCspRomTxdCARLviKw6E5SY8.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'github.com,192.30.253.113' (RSA) to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:SuperHouse/esp-open-rtos.git' into submodule path 'esp8266-proxy/esp-open-rtos' failed`

kanflo, 2018-03-09:
Bummer, I added the EOR submodule as git@github.com rather than https://github.com meaning it will fail if one does not have a public key uploaded to Github. Will change tonight (CET :).

jovandervyver, 2018-03-10:
Still looks broken, but in the meantime, those interested may want to take a look at:

[Creating SSH keys](https://www.digitalocean.com/community/tutorials/how-to-set-up-ssh-keys--2)
[Adding SSH keys to Github](https://help.github.com/articles/adding-a-new-ssh-key-to-your-github-account/)

jandcmoore, 2018-03-10:
Yes I did that last night - uploading a public ssh key allows clone just fine.

kanflo, 2018-03-10:
Last night got postponed ;)

closed 2018-03-10 by kanflo