kanflo/opendps #27: Past corruption for units smaller than 4 bytes

kanflo, 2018-04-13:
I did not note today is Friday the 13th until I found a bug in the past.c code. Writing a unit smaller than 4 bytes will corrupt the past. Guess I could have been worse off today :D

kanflo, 2019-01-01:
<4 bytes, %4 is handled IIRC

robots, 2025-04-13:
```
@@ -279,7 +279,7 @@ bool past_write_unit(past_t *past, past_id_t id, void *data, uint32_t length)
         if (!flash_write32(end_address+UNIT_SIZE_OFFSET, length)) {
             break;
         }
-        while(4*wi < length) {
+        while((length >= 4) && (4*wi < length)) {
             temp = ((uint32_t*)(data))[wi];
             if (!flash_write32(end_address+UNIT_DATA_OFFSET+4*wi, temp)) {
                 break;
```
I know you are not maintaining this project anymore,but if anyone is interested this is the fix for the problem :)

And of course you need to remove the check:
```
@@ -232,7 +232,7 @@ bool past_read_unit(past_t *past, past_id_t id, const void **data, uint32_t *len
 bool past_write_unit(past_t *past, past_id_t id, void *data, uint32_t length)
 {
-     if (length < 4) {
-        return false; /** https://github.com/kanflo/opendps/issues/27 */
-     }
     if (!past || !past->_valid || !data || !length || id == PAST_UNIT_ID_INVALID || id == PAST_UNIT_ID_END) {
 #ifdef DPS_EMULATOR
```