kanflo/opendps #31: Support for DPS3005

cjkoenig, 2018-05-05:
Hi,
I have a DPS3005 power supply and would love to know if it is supported by OpenDPS. At the moment, the only difference I see is the max. output current of 3A instead of 5A as DPS5005 has. Is it sufficient to just change the MAX_CURRENT to 3000 in opendps/Makefile?

kanflo, 2018-05-05:
In theory it should work unless RDTech did some hardware changes I have yet to learn. You could change the makefile but adding a parameter to make will work too:

```make MAX_CURRENT=3000```

Be brave and let me know how it worked out ;)

cjkoenig, 2018-05-06:
Thank you for the hint- today, I was brave and it worked out fine! 

Let me share my experiences I made during the flashing of my DPS3005 and how I fixed them:
- To run OCD you need root privileges (required two times: once during inital connect, then during flashing). Unfortunately, the error message is not clear that an access right to the USB flasher is missing...I ran openocd with sudo/modified the makefile with a sudo before the openocd call (maybe there is a more elegant way?)
- If you solder pins for flashing your DPS, remove them afterwards because the enclosure is not fitting correctly (if I understand correctly, an update can be done with UART anyway)
- To run dpsctrl.py for controlling the DPS with a serial connection, I recommend to create a virtual environment that provides a python2 interpreter:
-- `virtualenv -p /usr/bin/python2 venvs/python2`
-- activate it with `source venvs/python2/bin/activate`
-- and install the two required python packages with `pip install pyserial pycrc` (on my regular system with python3 I was not able to run dpsctrl.py)

Thank you for this project :+1: Now I have a tiny power supply on my desk and only wait for an ESP-01 board for remote control.

closed 2018-05-06 by cjkoenig

kanflo, 2018-05-13:
Glad to hear the '3005 is working! For OCD, add yourself to the ``dialout`` group and logout/login. I will update the docs with your finding.

kanflo, 2018-07-18:
Interesting UI bug you found there. I have never tried >12V supply which one could think would be enough to test "four digit voltage setting".

Do you mean you can upgrade using dpsctl but not change settings? Please note the switches to dpsctl changed a few commits ago.

Also, when you say "version 2.3", where did you get that information from? I have never seen any HW revision markings on my DPS:es.

Nannak, 2018-07-18:
Here is the version number printed to the back of the board with the connection points, right by the connector.

[image]

I did remember reading in a comment somewhere that someone posted a version number but couldn't find that again - I used the switches in the most recent posts I could find but will try again, also will try with a lower input voltage and see what I get! The 3005 isn't a booster though.

Nannak, 2018-07-18:
OK so simple as that, 10v input works fine! Well spotted!

Thank you! So I'll see what the threshold at which it starts to behave badly is...