kanflo/opendps #32: Output Voltage is not set if only remote control is used

pgunzelmann, 2018-05-10:
powering up the DPS5005 and try controll by dpsctl.py only, the output voltage seems not to be set. 

>python dpsctl.py -d com4 -v -u 1000
Communicating with com4
TX  7 bytes [7e 02 03 e8 47 15 7f]
RX 6 bytes [7e 82 01 6d db 7f]

>python dpsctl.py -d com4 -v -i 1000
Communicating with com4
TX  7 bytes [7e 03 03 e8 70 25 7f]
RX 6 bytes [7e 83 01 5e ea 7f]

>python dpsctl.py -d com4 -v -p 1
Communicating with com4
TX  6 bytes [7e 05 01 ef d4 7f]
RX 6 bytes [7e 85 01 f4 4c 7f]

>python dpsctl.py -d com4 -v -s
Communicating with com4
TX  5 bytes [7e 04 40 84 7f]
RX 17 bytes [7e 84 01 13 df 03 e8 00 00 00 00 03 e8 01 59 7b 7f]

V_in  : 5.08 V
V_set : 1.00 V
V_out : 0.00 V (on)
I_lim : 1.000 A
I_out : 0.000 A

also the new set values are not shown in the dps5005 display (may its a following issue)

kanflo, 2018-05-13:
That's too bad. I might have messed things up with the lashes CC work, could you try an earlier commit, eg. ``43f0eba``?

pgunzelmann, 2018-05-13:
Thanks for the hint. I can try. But bye the way I littele bit wondering why you have two different versions... For my understanding the setting values shuould be max current and max volltage.. And the connected resistance decide  which type of controll will applay.. 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit 1&1 Mail gesendet.

pgunzelmann, 2018-05-14:
Whit an earlier commit (43f0eba) i done the same procedure. Now the status querry show the expected results.

V_in  : 5.07 V
V_set : 1.00 V
V_out : 0.97 V (on)
I_lim : 1.000 A
I_out : 0.000 A

Thanks for quick response and good luck to clean up the "messed things" ;)

kanflo, 2018-05-14:
Glad it's working, sorry I broke things and above all thanks for reporting it. I will keep this issue open until Imend things.

kanflo, 2018-06-29:
Remote control is fixed now, please let me know if I was wrong.

closed 2018-06-29 by kanflo