kanflo/opendps #33: bin file for testing

dziedzic99, 2018-05-18:
Hi, i think I fail to compile opendps correctly.
To test that hipothesis - can someone please post a working opendps.bin file?

dziedzic99, 2018-05-18:
I manage to flash it according to command output, bootloader works, but the program itself doesn't. The screen doesn't light up and I can't talk to it over UART.

kanflo, 2018-05-18:
That's odd. Did you flash the boot loader and app in the correct order (boot loader, then app)? I noticed that OpenOCD overwrites the beginning of the app when flashing the boot loader.

kanflo, 2018-06-29:
Ping @dziedzic99. Did you have any success in flashing your unit?

dziedzic99, 2018-06-30:
Yup. I flashed it finally but only after I wiped my toolchain clean and
reinstalled it.

closed 2018-10-01 by kanflo