kanflo/opendps #34: Bootloader 176 bytes too large (GCC 8.1)

tracernz, 2018-05-22:
```
$ make -C dpsboot
make: Entering directory '/home/mike/dev/opendps/dpsboot'
  CC      common.c
  CC      hw.c
  CC      dpsboot.c
  LD      dpsboot.elf
/usr/lib/gcc/arm-none-eabi/8.1.0/../../../../arm-none-eabi/bin/ld: dpsboot.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/8.1.0/../../../../arm-none-eabi/bin/ld: section .app VMA [0000000008001400,000000000800f7ff] overlaps section .text VMA [0000000008000000,00000000080014a3]
/usr/lib/gcc/arm-none-eabi/8.1.0/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 176 bytes
collect2: error: ld returned 1 exit status
make: *** [../libopencm3.rules.mk:199: dpsboot.elf] Error 1
make: Leaving directory '/home/mike/dev/opendps/dpsboot'
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I managed to build it okay with GCC 6.3.1, but it's close too:
```
                0x0000000000001400                boot_size = 0x1400
...
.text           0x0000000008000000     0x1394
```

kanflo, 2018-05-22:
Thanks for reporting. You could try adding ``-flto`` to the ``CFLAGS`` which reduces the size of the BL quite significantly. I have not had time to test it but it should not introduce any side effects.

tracernz, 2018-05-22:
That indeed made it much smaller. :+1: 
`.text           0x0000000008000000     0x1264`

I haven't tested it to make sure it actually works yet.

CBJamo, 2018-06-06:
Adding the -flto flag breaks the bootloader, not sure why. For the moment, I butchered the bootloader by stubbing out the code in the handle_upgrade function. If anyone else does the same, bear in mind that you'll have to break out the programmer to flash updates.

kanflo, 2018-06-06:
Thanks for trying. I will dig into the bootloader later this summer when I have refactored the stuff I am working on right now. I recall the [past garbage collection](https://github.com/kanflo/opendps/blob/3a00c6a55354d06dd71c19ba1f7eb8434d759e97/opendps/past.c#L451) takes quite some space and could be omitted if the app promises to perform GC before invoking the BL.

konsgn, 2018-08-07:
For my fix, i just changed the boot_size to 6K in both the linker scripts of dps boot and opendps.

Don't these devices actually have the RB 128k flash versions that are commonly labled as R8 64k in china?

Xenoamor, 2018-08-12:
The -flto flag breaking the bootloader may well be related [to this](https://bugs.launchpad.net/gcc-arm-embedded/+bug/1747966)

I've been seeing this bug on 7-2017-q4-major and 7-2018-q2-update

Xenoamor, 2018-08-12:
I'm not sure how you got it to build with GCC 6.3.1...
This is under Ubuntu

```
$ make -C dpsboot
make: Entering directory '/home/user/Documents/opendps/dpsboot'
  LD      dpsboot.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: dpsboot.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: section .app VMA [0000000008001400,000000000800f7ff] overlaps section .text VMA [0000000008000000,000000000800142f]
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 60 bytes
collect2: error: ld returned 1 exit status
../libopencm3.rules.mk:198: recipe for target 'dpsboot.elf' failed
make: *** [dpsboot.elf] Error 1
make: Leaving directory '/home/user/Documents/opendps/dpsboot'
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

lmandjes, 2018-09-22:
I have exactly the same issues on ubuntu, did you find a solution?

lmandjes, 2018-09-22:
I see last line was using 7.2.1 version

lmandjes, 2018-09-22:
How do i install 7.2.1 on ubuntu? downloaded the tar mut do not know how to link it into the toolchain

Xenoamor, 2018-09-23:
Try an earlier version. I can't recall which worked for me though

kanflo, 2018-09-23:
The BL is not reduced by ~300 bytes which hopefully should be enough. I am reluctant to changing the BL size as this would break peoples DPS:es until they realised the BL needs refreshing.

closed 2018-10-01 by kanflo