kanflo/opendps #36: Voltage does not turn (goes back to zero) after pressing turn off button

Robin10155, 2018-06-28:
I seeing a weird issue where the output voltage does not turn off after pressing on/off button.
When the press the on/off button (after resetting opendps) the power turn on but it does not turn off (goes back to zero) after pressing button again (though the lcd shows power off). Voltage remains the same even if i set the voltage to something else. Is this the intended behaviour or is it a bug in the firmware?

kanflo, 2018-06-29:
That would definitely not be the intended behaviour ;) What DPS model do you have?

tzarc, 2018-07-13:
Aha, I should've read the github issues before posting on the blog.

Got the same issue with revision 10ae4b2cd41511770c260888928f26bdcdba0e2f.

From just before I flashed the firmware, with the manufacturer-provided firmware still installed:
5V-on.txt: https://pastebin.com/kdaw6pD5
5V-off.txt: https://pastebin.com/SDTb15nQ

After flashing opendps, taking a readout straight after reset, and after a power on/off cycle:
just-after-reset.txt: https://pastebin.com/73i8f4LD
after-power-on-power-off.txt: https://pastebin.com/JqAQ3nHL

kanflo, 2018-07-13:
Thanks, I will look into this.

tzarc, 2018-07-17:
I've forced-written all the GPIO registers using openocd to match the manufacturer "off" state, but I still get voltage output.
Wondering if perhaps there's some latching circuitry that's been added in a later revision of the device?

tzarc, 2018-07-17:
About the only thing I can do to get it to zero-out is forcefully setting `DAC_DHR8R1 = 0;` -- voltage output is still present though. Setting it to anything non-zero shows a response on the display.

I doubt this is the only mechanism they're using, despite the DAC values all being zero on register dumps from their firmware. Perhaps for sanity's sake opendps should be zero'ing out the DAC anyway.

tzarc, 2018-07-17:
Added PR #42 - this seems to function just fine on my DPS5005, even though I feel it's a bit iffy considering toggling PB11 seems to have no effect. Either way, for safety this looks like it should be okay?

geekbozu, 2018-08-01:
Just to add to this, Mine does not turn off after a CV current over limit event either. Have not tested said PR yet however.

Xenoamor, 2018-08-12:
Can confirm that the power does not switch off
`Compiled with: arm-none-eabi-gcc (GNU MCU Eclipse ARM Embedded GCC, 64-bits) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]`

jimmyw, 2018-08-29:
Try PR #42, it solved the issue for me.

f0086, 2018-09-28:
Same problem here.
Is #42 save to use? Looks like this is still WIP?

kanflo, 2018-10-01:
I managed to reproduce the issue on a newer '5005 and #42 fixed it. I recall adding DAC zeroing some time ago but it got lost in the manual merge from the CC branch.

closed 2018-10-01 by kanflo