kanflo/opendps #41: Overvoltage limit protection in CC mode

Robin10155, 2018-07-13:
Can we add an overvoltage limit in constant current mode?

kanflo, 2018-07-18:
That would be doable. I just need to figure out how it would work out UI wise.

Robin10155, 2018-07-20:
Constant voltage mode is doing something similar already. We set constant voltage value using voltage setting and current limit using current setting. Similarly in constant current we can set constant current value using current setting and voltage limit using voltage setting.

jimmyw, 2018-09-01:
@Robin10155 Did a small redesign, and added this feature, se my branch: https://github.com/jimmyw/opendps

piit79, 2018-12-10:
@jimmyw Could you please post a picture of your modified UI? :-)

Xenoamor, 2019-02-12:
I'll have a look at implementing this soon.
I think adding a `handle_ovp()` function into the `adc1_2_isr` would be the best way to do this. This limit can then be set with `pwrctl_set_vlimit()`

PR #127 (2019-02-17): Add overvoltage protection to CC mode
This is to close #41. This required quite a few additions but it's handled in the ADC ISR like the OCP so it should be nice and fast
closed 2019-02-17 by kanflo