kanflo/opendps #43: Bluetooth Module

RLovelett, 2018-07-23:
I am interested in providing support for the bluetooth module that comes with the newer DPS. With the ultimate goal of eventually making OpenDPS available via an iOS application using Bluetooth.

As far as I can tell, the Bluetooth module just appears as a serial device.

I guess my questions are:

1. Do you think this sounds possible?
2. Has anyone attempted to make use of the bluetooth module?

kanflo, 2018-07-23:
The last time I checked (this was ~9 years ago :) iOS did not support BT serial ports but hopefully this is not the case anymore. I am not aware of anyone using the BT module but depending on Apple, your idea is either easy peasy or completely impossible. Either way, it's a cool idea!

RLovelett, 2018-07-23:
Yeah, I think after searching and just playing around. The _iOS_ part is _completely impossible_. 👎

Two paths seem available:

1. Replace the OEM Bluetooth board with a BLE board. BLE is easily hackable inside of iOS.

The bad news is I cannot make use of the bluetooth module that it comes with. Though this is not really a deal breaker.

2. Retreat to macOS where it is possible.

The bad news here is that I was really hoping to make it to a mobile device (that I already own).

Presumably the ESP8266 in your example is talking directly with the OpenDPS firmware. Is there any documentation on how these two devices communicate?

kanflo, 2018-08-25:
The protocol is described [here](https://github.com/kanflo/opendps/blob/master/opendps/protocol.h), feel free to holler of something is unclear. The ESP8266 is just a proxy (what's why the firmware is called esp8266-proxy ;) and will tunnel the frames from ```dpsctl.py``` unmodified to the OpenDPS firmware. Any response from the DPS is sent back from where the previous frame originated from.

closed 2018-10-01 by kanflo