kanflo/opendps #45: Release pre-compiled builds?

ricktendo, 2018-08-20:
Any chance you could release pre-compiled binaries for the different supported models and publish these in the "Releases" section?

I have a DPS5015 and a ST-LINK/V2 flasher but not much experience compiling the builds.

DirtyEngineer, 2018-08-22:
A little over a year ago the pre-compiled binaries[ were removed](https://github.com/kanflo/opendps/commit/b14dc7b1435a4cf74628d5cf7d6eac118127d666). The reason given was that "Time does not permit compiling and releasing the STM32 binaries, especially since building these are very easy given the ARM support of gcc." ([Source](https://johan.kanflo.com/upgrading-your-dps5005/#comment-7232)). I know at one time there was talk of creating a more detailed walk through of compiling and flashing.

Xenoamor, 2018-08-24:
I had to compile it with the `7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]` version. Above this appeared to fail

zoobab, 2018-08-25:
I tried to make a Dockerfile inspired by the one I use to compile DirtyJTAG. End result is that it does not build. So I would welcome released binaries as well.

Here is my Dockerfile attempt:

```
FROM debian:jessie-slim
LABEL Description="OpenDPS firmware for stm32" 

RUN DEBIAN_FRONTEND=noninteractive apt-get update -y -q && apt-get install -y -q sudo make python gcc-arm-none-eabi git-core libnewlib-arm-none-eabi

ENV user opendps
RUN useradd -d /home/$user -m -s /bin/bash $user
RUN echo "$user ALL=(ALL) NOPASSWD:ALL" > /etc/sudoers.d/$user
RUN chmod 0440 /etc/sudoers.d/$user

USER $user
WORKDIR /home/$user
RUN mkdir -pv code
COPY . ./code/
RUN sudo chown $user.$user -R /home/$user/code
WORKDIR /home/$user/code/
RUN git submodule init
RUN git submodule update
RUN make -C libopencm3
RUN make -C opendps flash
RUN make -C dpsboot flash
```

Then I hit this error:

```
  CC      tick.c
  CC      tft.c
  CC      spi_driver.c
  CC      ringbuf.c
  CC      ili9163c.c
  CC      dbg_printf.c
  CC      mini-printf.c
mini-printf.c: In function '_putc':
mini-printf.c:107:17: warning: declaration of 'ch' shadows a previous local [-Wshadow]
  int _putc(char ch)
                 ^
mini-printf.c:105:7: warning: shadowed declaration is here [-Wshadow]
  char ch;
       ^
  CC      font-0.c
  CC      font-1.c
  CC      func_cc.c
  CC      uframe.c
  CC      protocol.c
  CC      protocol_handler.c
  CC      opendps.c
  LD      opendps.elf
stm32f100_app.ld:15: nonconstant expression for length
collect2: error: ld returned 1 exit status
make: *** [opendps.elf] Error 1
../libopencm3.rules.mk:198: recipe for target 'opendps.elf' failed
make: Leaving directory '/home/opendps/code/opendps'
The command '/bin/sh -c make -C opendps flash' returned a non-zero code: 2
```

kallemooo, 2018-08-25:
Hi

The problem is that the Ubuntu and Debian package gcc-arm-none-eabi that provides GCC cross compiler for ARM Cortex-A/R/M processors is to old and is not usable.

Install the official toolchain from ARM and use that instead.
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

Also see issue https://github.com/kanflo/opendps/issues/22

kanflo, 2018-08-25:
Thanks for pointing out the compiler weirdness @kallemooo. I will add it to the documentation. Pre compiled binaries will arrive the day I get something Jenkins like up and running. Precompiled _untested_ binaries that is as I only have ´5005 to test with.

ricktendo, 2018-08-25:
Hey Jordan,

Not that I have the money but would you be open to people donating other DPS units so you can have to test/develop for?

Ric

goosenphil, 2018-08-26:
I've modified @zoobab 's Dockerfile to now build it with GCC 7.1.0 (by using Fedora 28) and it now compiles successfully. Another possible reason why the build originally might have failed is that ```make
 flash``` expects the STM32 programmer to be present. Instead now the Dockerfile builds the firmware with the image and you can just run the container to get the binary (or flash it).

Here is the [Dockerfile](https://gist.github.com/goosenphil/1e29b44e0222a134451468046bcdcb39)

To use it, just clone the OpenDPS repo, then cd into it, place the "Dockerfile" and then run docker build.
Example:
```bash
git clone https://github.com/kanflo/opendps.git
cd opendps
wget 'https://gist.githubusercontent.com/goosenphil/1e29b44e0222a134451468046bcdcb39/raw/7cc8044f861406fff033636edb6c5f18aaaffe10/Dockerfile'
docker build -t="goosenphil/opendps" .
```

goosenphil, 2018-09-01:
@zoobab  Done! Could you please have a look at it here? https://github.com/kanflo/opendps/pull/47

goosenphil, 2018-09-01:
@zoobab Thanks for sharing the script, though I had permission issues with the files it copied over so I've taken a different approach: Instead of running the container, it just makes one from the image and then copies the files over (solving the permission issues by using docker cp). At the end I delete the container.

zoobab, 2018-09-02:
What is Dockerfile3?

ricktendo, 2018-10-29:
I would like to compile the firmware for my DPS5015, can somebody update the readme with the instructions or give me an example of how I compile the opendps and dpsboot for the DPS5015?

I found this but I am uncertain what max current I should really use
`make -C opendps MODEL=DPS5015 MAX_CURRENT=9999`
Also, can somebody tell me if the dpsboot is model specific or is this same for all models?

ricktendo, 2018-11-01:
Thanks kanflo I was able to compile but I did receive many, many warnings

I have a DPS5015, this is 15A so would this be correct?

`make -C opendps MODEL=DPS5015 MAX_CURRENT=14999`

If I leave out the **MAX_CURRENT** command, will it compile using the default maximum current rating of the model?

kanflo, 2018-11-01:
Yes, that is correct. What warnings do you get and what version of ARM gcc are you running?

ricktendo, 2018-11-01:
1- I am using Ubuntu 18.04 on Linux Subsystem for Windows 10, when I used **gcc-arm-none-eabi-7-2018-q2-update-linux.tar.bz2** I get the [following warnings](https://pastebin.com/bYGW9bLE) when compiling **libopencm3**, none for **dpsboot**, and [one warning](https://pastebin.com/enkRki8i) for **opendps**

2- I also tried with `apt install gcc-arm-none-eabi` and I got **NO WARNINGS** when compiling **libopencm3** and none for **dpsboot**. But I did get [two warnings](https://pastebin.com/N90kq8HX) when I compiled **opendps**

kanflo, 2018-11-01:
Ah, I thought you got lots of warning from building opendps. Glad that was not the case ;)

ricktendo, 2018-11-01:
Are the warnings OK, should I ignore?

Just to confirm, it is OK if I leave out the MAX_CURRENT command because specified MODEL will use the default max current?

kanflo, 2018-11-01:
Yes, you can safely ignore the warnings. You will need to set MAX_CURRENT as it defaults to 5000. This is actually something that should go into [dps-model.h](https://github.com/kanflo/opendps/blob/master/opendps/dps-model.h) when I think about it.

ricktendo, 2018-11-02:
Please include default max current in dps-model.h, if somebody wants to override they can use the MAX_CURRENT command.

P.S. what are the offsets of the dpsboot and opendps, in case I wanted to flash manually?

kanflo, 2018-11-12:
@ricktendo : 0x08000000 for the bootloader and 0x08001400 for the app. Max currents in dis-model.h is a good idea.

Xenoamor, 2019-01-07:
I've noticed Travis CI is free and fairly easy to integrate with Github. They run the server for you as well so you don't even need to setup your own server

kanflo, 2019-01-07:
Thanks @Xenoamor, I will definitely have a look.

JackNewman12, 2019-09-16:
Since travis will now automatically add the binaries to any commits that are tagged. Should we close this ticket?

Also we can now add a [image] to the README

zoobab, 2019-09-16:
A quick comment: you can frankly move most of the build.sh logic inside the Dockerfile:

https://github.com/kanflo/opendps/blob/master/docker/build.sh

And use an entrypoint to copy the binary from the container to the host.

kanflo, 2019-09-19:
Closing due to #154 
Tags are built as well as the latest and greatest on master. Kudos to @danielkucera.

closed 2019-09-19 by kanflo