kanflo/opendps #48: Problem setting voltage with CV mode

007trains, 2018-09-05:
I have flashed the dps5005 unit successfully and tested cc mode which works ok (setting 1A results in 1.6A running). Then i tried cv mode but when i go to set the voltage it jumps from 0.00 to 17.46 then 17.63 then 17.80. I cannot set the voltage correctly, tried rebuilding/reflashing without success. Not sure if this issue is just me or the version i have compiled

kanflo, 2018-09-23:
I would guess the DPS hardware changed. It's neither you nor your setup ;) I will see if I can reproduce it on one of my newer DPS:es.

f0086, 2018-09-28:
I have a similar problem. If I set the Voltage to 5V and turn it on, it measures ~4.7V. Same for other voltages (no load attached).

kanflo, 2018-10-01:
A calibration mode has been discussed some time ago and probably should be implemented. Still this variation is unexpected of the hardware unless the design changed.

kanflo, 2018-10-01:
I see exactly the same behaviour wrt. output voltage on my new '5005. I will look into solving this at build time.

kanflo, 2018-10-01:
As this is a duplicate of #17 I am closing this one.

closed 2018-10-01 by kanflo