kanflo/opendps #50: DPS5015 setting in dps-model.h does not work 100% with the latest hardware

wangxiatao, 2018-09-23:
Hi, I bought a dps5015 fairly recently from a distributor called "KKmoon" from Amazon EU, and I quickly put in opendps after I did some testing on the stock firmware.

Two problems as I can tell right now. First off the settings in dps-model.h does not work correctly on the hardware I have (on the PCB it says version *V2.3*):
* ADC on the voltage reading seems to be OK
* DAC on the voltage is off by about 5%, i.e when setting 12V, it outputs about 11.4V
* ADC on the current fluctuates quite a lot (when the mean is about 200mA, it displays between 100 and 300mA), and the mean number is not accurate either (it's hard to say what exactly the mean is since it jumps up and down, but if I guess it'll be off by 50%
* DAC on the current is hard to measure since I yet to find a way to turn it into CC mode on the UI, and I have not tried serial or wifi yet

The second issue is the fan, somehow the fan is slowing ramping up speed regardless of what kind of load is on the output, it feels there's some DAC/PWM output drifted upwards and causing the power supply to the fan to go up. It does turn off if I press the power button to disable output.

Saying that, for the first one, at least on the voltage DAC, I can give it a try and measure the numbers then refine the dps-model.h, as long as I know where to measure, any guidance on that?

Thanks for all your answers in advance!

/Tao

wangxiatao, 2018-09-23:
Thanks for getting back to me so quickly Johan! Yeah...well, I missed that step to dump registers, what a mistake :(

I didn't have the same issue as some people have with the on/off button, so I guess the hardware change isn't that substantial, however I only have this dps so it's pretty hard for me to compare. Regarding the CC mode, I don't know if I missed any steps but I don't see it the way you described, I'd guess the UI will show "CC" if it's in CC mode? I did this:
1. Boot up and wait till "wifi" stops flashing, by default it boots to CV mode
2. Press "SET", and the cursor defaults to voltage
3. Press "M2", run the knob and set the current
4. Press "SET" again to confirm, the cursor disappears and the parameters are set (I guess)
5. Press "ON/OFF"
6. UI is still showing "CV" and the power is still **cut off** when current is over the set limit

ricktendo, 2018-09-28:
> 
> 
> BTW, will it help if I buy a new one from the same source I bought earlier on, and ship over to you? So you have a hopefully identical dps as I or other people have, with stock firmware to measure on. I'd guess a lot of people will go for 5015 simply because it's quite comparable price as 5005, and given it's active heat sink it should do a better job on heavy load. At least that what I thought when buying it, I can see the sales of 5015 is going to be up.

Did kanflo accept your generous offer to provide a 5015 for development?

I ask because I also have a 5015 and would love for this model to be officially supported!

kanflo, 2018-10-01:
If you generously send me a '5015 @wangxiatao I would be happy to look at it. I have two other pressing matters on my todo list though, one just got covered on Hackaday, so I would guess "November" on '5015 support.

wangxiatao, 2018-10-10:
> Did kanflo accept your generous offer to provide a 5015 for development?
>
> I ask because I also have a 5015 and would love for this model to be officially supported!

Yes he did :) According to @kanflo good things might happen in November, stay tuned

u22br, 2018-10-13:
Hi,

I have DPS5015 PCB revision V2.3 (bought from TomTop an year ago).
I did register dump before reflashing it to OpenDPS.
[DPS5015_debug_data.tar.gz](https://github.com/kanflo/opendps/files/2475882/DPS5015_debug_data.tar.gz)
After reflashing it does not working properly, but I guess I can make mistake somewhere.
Before reflashing I desoldered STM32F100 and soldered new one instead of it, 
so I have backup chip with stock firmware.

u22br, 2018-10-17:
Hi,

Today I found the reason why my OpenDPS setup was not worked.
I wasted a lot of time trying to found where I made a mistake
when I edited source code files/compiled a project/flashed it to mc/something else.
Finally I decided to check microcontroller ID and bingo :)
As I said before, I desoldered original STM32F100C8T6B for have a backup and soldered 
other MC which I bought on Aliexpress 3 month ago from this seller:
https://ru.aliexpress.com/item/5PCS-STM32F100C8T6B-QPF32-MCU-new-and-original-IC-free-shipping/32461309300.html

Unfortunately I have got STM32F100C8T6 (without "B" at the end) but this is not so important
more important that this is supposed to be remarked STM32F103.
What said openocd about this remarked chip:
root@stm32-dev:~/opt/xPacks/@gnu-mcu-eclipse/openocd/0.10.0-8.1/.content/scripts# openocd -f interface/stlink.cfg -f target/stm32f1x.cfg -c "init" -c "$target_name mdw 0xE0042000 1" -c "exit"
0xe0042000: **20036410**

information from reference manual for STM32F103xx
[image]

What said openocd about original chip from DPS5015:
root@stm32-dev:~/opt/xPacks/@gnu-mcu-eclipse/openocd/0.10.0-8.1/.content/scripts# openocd -f interface/stlink.cfg -f target/stm32f1x.cfg -c "init" -c "$target_name mdw 0xE0042000 1" -c "exit"
0xe0042000: **10016420**

information from reference manual for STM32F100xx
[image]

photo of remarked chip
[image]

wangxiatao, 2018-10-17:
Aha so this is how a register dump looks like! How does one make use of this dump?

I just checked my stock MCU and it has the same response as your original MCU, just to confirm that. Now knowing this, have you been able to get your dps to work @u22br ?

u22br, 2018-10-17:
Today I ordered STM32F100C8T6B from local distributor, hope that I will get it in 2-3 days.
I guess that my DPS will work OK with right stm32 because of with fake STM32F100 everything was fine except of reaction on ON/OFF button. That is due to STM32F103 does not have DAC.

u22br, 2018-10-22:
Hi,

Some updates.
With genuine STM32F100C8T6B my dps has appropriate reaction on ON/OFF button, output voltage appeared but there is some issue as wangxiatao described in first message. Output voltage a bit less than I set up, for example if I set 5V real output voltage is about 4.68V.
Also recently I have got new toy - DPS5020. It also has PCB revision V2.3 and I took a photos of all active components on both DPS5015 and DPS5020. I am not sure that this issue thread is right place for posting these photos but if @kanflo confirm that it is OK, I will post it here.
In my opinion the difference between 5015 and 5020 very little - output choke, one electrolytic capacitor, input mosfets.

kanflo, 2018-10-22:
Please feel free to post the photos @u22br.

kanflo, 2018-10-22:
> Aha so this is how a register dump looks like! How does one make use of this dump?

@wangxiatao The register dump can be used to check what state of the STM32 peripherals (DAC, timers, GPIOs, ...) when the original firmware is running. With this information it is possible to check if the original firmware uses a peripheral that OpenDPS currently not supports.

u22br, 2018-10-22:
later I will edit this post and add to it some additional info and photos
.
.
**5015**
[image]
[image]
all four output mosfets same as this one:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
.
.
.
.
**5020**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

wangxiatao, 2018-10-23:
Very nice pictures! Can you see which component is the current shunt? I'm interested to know which ADC pins are used by the ST micro to measure current.

I'm not able to use "try an error" to fix the current, I know it's silly and I should've started with measuring the right components, but I actually did that on the voltage DAC and the value below gives me good setting on the voltage, which is confirmed by another multimeter:
`#define V_DAC_K (double)0.076700`

u22br, 2018-10-23:
@wangxiatao sorry, I am not sure that I correctly understand what you mean when say "which component is the current shunt". There are three current shunts on the board near output terminals. Near it (in fact right under current shunts) there are four active components (all sot-23):
SOT-23-5 | marking on the board: U6 | marking code: A63A  | LM321M
SOT-23-5 | marking on the board: U3 | marking code: S0BFA  | SGM8581 regarding to this information https://www.elektroda.pl/rtvforum/topic3442107.html
SOT-23 | marking on the board: Q02 | marking code: Y1  | SS8050
SOT-23 | marking on the board: Q6 | marking code: 702 | 2N7002
(I can not sure that marking codes corresponds to the specified components, please double check it)

[image]
[image]
[image]
[image]

u22br, 2018-10-24:
Hi,

Today I decided to do a big job, I desoldered almost all components from pcb which cover pcb traces.
So with these photos everyone can find out connections between components.
And may be someone will be brave enough to do schematics using these photos :)
Please note that I will solder pcb back nearest days, so I can do more photos if something hard to examine on provided photos.

[image]
[image]

wangxiatao, 2018-10-25:
@u22br sorry I should've described my question better, I was looking for the amplification output of the voltage drop over the current shunts. The components you mentioned are probably for amplification, so it's interesting to know which pin is the output of the final signal before it goes into the ADC.

The current shunts have extremely low resistance as I measured in room temperature, I don't know if it changes with the current which brings up the temperature. I'm also using quite poor equipment so the measurement is probably below the multimeter's resolution, I only read 0.1ohm to 0ohm. Again I know very little about electronics, maybe this revised description is again incomprehensible :)

u22br, 2018-10-25:
@wangxiatao 
I guess that interested your path goes something like that (maybe I'm wrong):
[image]

Jantek, 2018-10-26:
Hi everybody! I would like to report the strange behavior I see on my new DPS5015 and DPS5020 (both V2.3). Of course I have no register dump... **stupid, I know!**

When I am in the CV-mode I can't setup the voltage. As you can see here in the pictures the voltage jumps between 0V and approximately 83V. Even if I am at the last digit. Sometimes the PSU crashes and need to be restarted, sometimes the LCD gets completely white. The setup of the current seems to work. Also in the CC-mode.

[image]
[image]

DPS5015 and DPS5020 seem to act the same to me. I have compiled the source with ```make -C opendps MODEL=DPS5015 MAX_CURRENT=9999```

Jantek, 2018-10-26:
I should have read the other issues first! Isn't it #48 with have been closed for #17 ?

gnbl, 2018-11-29:
I have used @u22br 's scans from https://github.com/kanflo/opendps/issues/50#issuecomment-432616154 to trace the bottom layer for an overlay:
[image]
Note that there are separate bottom fills and some vias connect to these.

gnbl, 2018-11-30:
@u22br Could you get the values for RT1, R29, RL? I suspect these could be used for temperature sensing - not sure if U7 has a part in it. Also present on DPH5005.

u22br, 2018-11-30:
Hi
@gnbl 
right now I can measure it only without de-soldering
if it is not OK for you, please let me know 
R29 - 9.63kOhm (slightly increased during measurement, may be actually it close to 10kOhm)
RL - 10.0kOhm
RT1 - 8.97kOhm (slightly increased during measurement, may be actually it close to 10kOhm)

gnbl, 2018-11-30:
@u22br Thanks, I got very similar values - posted in a comment at https://johan.kanflo.com/the-future-of-opendps/ with other findings, but it's not published yet. I suspect U7 is a voltage regulator or reference, but I have not yet found pin compatible parts (assumption: 1 = EN, 2=GND, 3=Out, 4=Vcc, 5=GND). EN then would be controlled by PD1, PA0 would be an ADC to measure the reference voltage, which also sources RL-R29+RT1-GND. PA6 then would measure across RT1. This then is a 10k NTC - pulling PA6 to GND causes the fan to start.

gnbl, 2018-12-02:
Although my comment since has appeared in kanflo's blog, I'll take the liberty to "dump" it here as well with minor modifications and additions:

I just got a DPS5015 and DPH5005.
There seems to be surprisingly little hardware info on these RD DPS/ DPH power supply modules out there.
So far I found 
https://github.com/kanflo/opendps/tree/master/hardware/reverse-engineering/dps3003 (KiCAD schematics)
https://circuitmaker.com/Projects/Details/Damian-Thompson-2/DPH3205-Reverse-Engineer (proprietary schematics)
https://github.com/kanflo/opendps/issues/50 DPS5015 PCB photos

Hardware info in the code
https://github.com/kanflo/opendps/blob/master/opendps/hw.c#L383
https://github.com/kanflo/opendps/blob/master/opendps/hw.h
https://github.com/kanflo/opendps/blob/master/ocd-client.py#L64

which means that there seem to be a few pins unused.

Microcontroller STM32F100C8T6B (C=48 pins, 8=64 Kbytes of Flash memory, T=LQFP -> medium-
density device, LQFP48, 8 Kbytes of SRAM)
Product page: https://www.st.com/en/microcontrollers/stm32f100c8.html
Datasheet: https://www.st.com/resource/en/datasheet/stm32f100c8.pdf
Reference Manual: https://www.st.com/resource/en/reference_manual/cd00246267.pdf
Errata: https://www.st.com/resource/en/errata_sheet/cd00260217.pdf

GPIO Pin usage on MCU, according to sources above
```
 1 VBAT
 2 PC13 power control on '5015
 3 PC14 NOT CONNECTED (DPS5015)
 4 PC15 NOT CONNECTED (DPS5015)
 5 OSC_IN NOT CONNECTED (DPS5015)
 6 OSC_OUT / PD1 -> U7.1 (SOT23-5, "M2Y8")
 7 NRST NOT CONNECTED (DPS5015)
 8 V_SSA -> U7.5 (DPS5015)
 9 VCC
10 PA0 (RL, U7.3 on DPS5015 and DHP5005)
11 PA1 button M2
12 PA2 button SEL
13 PA3 button M1
14 PA4 DAC1 V_set
15 PA5 DAC2 I_set
16 PA6 NOT CONNECTED (temp sensing on DPS5015?)
17 PA7 ADC V_out
18 PB0 ADC V_in
19 PB1 ADC I_out
20 PB2 BOOT1-GND (NOT CONNECTED on DPS5015)
21 PB10 NOT CONNECTED
22 PB11 power control on '5005 / fan control on '5015
23 GND
24 VCC
25 PB12 TFT Reset (P4.4 on DPS5015 v2.3)
26 PB13 TFT SPI SCK (P4.5 on DPS5015 v2.3)
27 PB14 TFT CS (P4.6 on DPS5015 v2.3)
28 PB15 TFT SPI MOSI (P4.7 on DPS5015 v2.3)
29 PA8 [TFT, unused?] (P4.8 on DPS5015 v2.3)
30 PA9 TX
31 PA10 RX
32 PA11 NOT CONNECTED
33 PA12 NOT CONNECTED
34 PA13 (SWDIO)
35 GND
36 VCC
37 PA14 (SWCLK)
38 PA15 not used in opendps? connected to current sense shunt and amplifier
39 PB3 not used in opendps? connected to current sense shunt and amplifier
40 PB4 button power
41 PB5 button rotary encoder
42 PB6 NOT CONNECTED
43 PB7 TFT backlight
44 BOOT0 (pull-down R12 on DPS5015)
45 PB8 rotary encoder
[... 126 lines cut ...]
GPIOx_ODR    = 0x0c

Confirmed with script output:
```
>>>> gpiod
GPIOD settings
...
ODR      : 0x00000002  [0x4001140c]
```

So we just need to write zero to 0x4001140c:
`opendps$ ./ocd-client.py w 0x4001140c 0`

Despite PD1 going low, the suspected output U7.3 does not change (3.3 V).

PA0 could be used for a reference measurement, as the LQFP48 package does not have VREF but probably uses VDDA as ADC reference.

Also, I have some "register dumps":
[DPS5015_HW2.3_FW1.6_5V1A_off_fan_on.txt](https://github.com/kanflo/opendps/files/2636044/DPS5015_HW2.3_FW1.6_5V1A_off_fan_on.txt)
[DPS5015_HW2.3_FW1.6_10V1A_on.txt](https://github.com/kanflo/opendps/files/2636045/DPS5015_HW2.3_FW1.6_10V1A_on.txt)
[DPS5015_HW2.3_FW1.6_5V_off.txt](https://github.com/kanflo/opendps/files/2636046/DPS5015_HW2.3_FW1.6_5V_off.txt)
[DPS5015_HW2.3_FW1.6_5V1A_on.txt](https://github.com/kanflo/opendps/files/2636894/DPS5015_HW2.3_FW1.6_5V1A_on.txt)

And a view with the CPU pinout overlay:
[image]

gnbl, 2018-12-02:
Even if U7 was supposed to be used as voltage reference, the stock firmware does not use it to calibrate the ADC. If the supply voltage drops, the module input voltage reading is much higher (19 V for 12 V at 1.9V VCC).

kanflo, 2018-12-14:
I am now looking at the DPS5015 that @wangxiatao kindly sent me. For the DPS5015 folks out there, what output is generally needed to make the stock firmware turn the fan on? I can control it as expected via PB11 but if there are ways of setting the speed I would like to get register dumps before wiping the stock firmware.

DavidMez, 2019-02-19:
Is it safe to work with DPS5015 now?

kanflo, 2019-02-20:
I am somewhat unsure to be honest. I have a '5015 but have not reflashed it yet as I am waiting for someone to tell me when the fan should be turned on so this behaviour can be mimicked by OpenDPS. I have no setup for pushing lots of power out of this thing.

gnbl, 2019-02-21:
@kanflo You could inject "temperature" voltages on PA6 to determine the threshold, or, if PWM is used on PB11, maybe other parameters. The register dumps I posted might show if PWM is configured (so would an oscilloscope).

sq9fk, 2019-03-04:
@kanflo 
Hello Johan,
Yesterday I've finnished assembly 1/2 DPS5015 units that I've bought last year. Flashed them with opendps firmware. Few things need to be mentioned:

PCB Rev. 2.2 (not 2.3)

1. LCD Freeze Issue.
After flashing from master branch this issue exist. Proposed solution in:
https://github.com/kanflo/opendps/commit/a88926ce2f2887f360f9f097da02162fa58bc4d8
Fixed this issue completely. Unit is stable (48h working). This should be merged in master branch

2. Calibration procedure from dpsctl - failed
Values after calibration procedure were wrong and units displays wrong values. I was forced to use Your google Excel Sheet and manual prepare the calibration values. I will upload them in the evening.

3. ADC is noisy.
Measurment is not stable +-20mV. I've searched for filter routine without lack... Maybe we should make simple filter AVG-ing /16 samples?

4. esp-proxy / esp-link
I finished the project with esp-link on esp8266. Could you clarify:
- esp8266-proxy - cannot reconfigure SSID without reflashing the unit? esp-link has nice Web UI to make all configuration. 
- esp8266-proxy - Lack of dual Soft-AP/Client mode. This cause difficult in field use when you don't know creditionals of the network.
esp-link needs driver like Com2TCP rather UDP scanning but is more configure friendly

5. CC mode is implemented? when set I < than load needs, executes OCP, rather than CC mode, or maybe I miss something? This is why I haven't done I_DAC callibration fo CC.

It's great project, I will try to contribute from my site! Thanks!

kanflo, 2019-03-04:
Hi @sq9fk. 

1. Thanks, I'll make sure it gets merged.

2. That's odd. Any idea why this would fail @Xenoamor?

3. Does that apply for all ADC channels?

4. Yes, the ESP Open RTOS based '8266 proxy lacks this functionality for now meaning it's quite cumbersome to change SSID.

5. CC mode is implemented but (probably) untested on the '5015.

Thanks!

sq9fk, 2019-03-04:
@kanflo 
I've just updated my previous post with calibration offsets, Maybe you can merge it into master for DPS5015 ? According to first posts in this thread I have the same problem with 4.74V after flashing with master. After cal: 5.00 (+0.03 peak because of ADC noise)

3. Yes. It's oscilates 2 or 3 ADC points. Simple filter will do the job in my opinnion but need to test it.
V_ADC and A_ADC.

5. Great I will test it in next days.

sq9fk, 2019-03-14:
About the noise:
There is nice project on github called Cheali-Charger (alternative firmware for LiPo etc. chargers).
They implement PID and Noise filtering.
https://github.com/stawel/cheali-charger/blob/master/src/hardware/atmega32/generic/50W/AnalogInputsADC.cpp

Firmware is now stable, it's for AVR family but we could implement similar algorithm.
What do you think @kanflo 

I've also add new issue about UI screen for current > 10A
https://github.com/kanflo/opendps/issues/138

kanflo, 2019-03-17:
Thanks for the link. We could start with a simple averaging algorithm and see how far that takes us.

polihedron, 2019-10-16:
My DPS5015 rev.2.3 always freeze LCD, when I touch few times multimeter probe to the ground connector. Please merge a88926c in master branch, I lost many hours to figure out, how to fix that..

Xenoamor, 2019-10-16:
I'll just tag @kanflo for visibility

kanflo, 2019-10-16:
@polihedron Is it the "unfreeze LCD" removal, the "TFT_CSN_PORT" addition or both that is needed? A PR is always welcome and will move faster to master ;)

polihedron, 2019-10-17:
@kanflo, "TFT_CSN_PORT" fixes strange DPS5015 LCD freeze when I touch positive multimeter probe to GND DPS5015 connector, LCD goes almost always white.  At first I suspected bad grounding ILI9163C issue, LCD connector loose, or LCD wires too long in noisy enviroment. 
Regarding to "TEMPORARY HACK TO UNFREEZE LCD" I added those two lines, because of strange color change, after some time it changes from white to dark yellow, or red with noise around edges, like maybe screen refresh was not working properly.
Still testing, but now I see huge difference in stability, LCD is stable as in DPS5005.
I will prepare PR.

kanflo, 2019-10-17:
It seems they decided to use a GPIO pin on the STM32 as slave select for the SPI driven display. Previously they simply grounded the signal on the display so I can see the display going haywire when the SS pin was left floating ;)

emilio-a, 2019-12-09:
Hello! I'm emilio, electronics engineer from San Juan, Argentina. First time as a github'er. I like and support OpenSource hardware and software, and also i'd like to contribute to this project. I will try to merge the two "big" photos from @u22br  into a single dual-layer image with different transparencies in GIMP, and then try to dump that into a kicad schem. If anyone has already done that, please tell me so I can contribute instead.
A simple non-commited google has revealed no schem for DPS5015 & DPS5020...at least I haven't found easily one.
Any comments, guides, helps or otherwise will be appretiated!

Xenoamor, 2019-12-09:
Hi @emilio-a, I'm not familiar with reverse engineering PCBs but your approach sounds like a good one. As far as I'm aware these are two layer boards

There's a rough schematic (not perfect) of [the dps3003 here](https://github.com/kanflo/opendps/tree/master/hardware/reverse-engineering/dps3003). I believe a large portion of the DPS5015 will be similar

kanflo, 2019-12-09:
Welcome to Github @emilio-a! Looking forward to see what you can come up with.

emilio-a, 2019-12-17:
yes , @Xenoamor, thanks! They are both very alike, but there are some interesting differencies. Ok, I am about to finish the 0.1 alpha version of the schematic...does anybodie know if I would be breaking any law by doing inverse eng and posting the schem here? I wouldn't like being sued around the globe...but it would be nice to contribute and  see how far can we go!

Xenoamor, 2019-12-17:
You'll need to check with your countries laws on the matter I'm afraid. I _believe_ it is okay in the US if you do it for the purposes of allowing it to work with something else (our software) and you aren't doing it for financial gain

ilium007, 2020-05-27:
Any update on the DPS5015? I just flashed mine before reading this whole thread :(

Now my DPS5015 outputs no voltage. For example, I can select 3.3v on the screen and then when I enable output I see it go to 0.03v and there is nothing on the multimeter off the outputs.

ilium007, 2020-05-27:
Noticed now also that when the output is off the internal fan turns on, when output is enabled (no voltage as per above) the fan turns off.
Suffice to say its bricked :(

emilio-a, 2020-05-27:
Ok, this is where I have got so far. It's still incomplete, and there are many tracks I can't figure out where are connected just from photographs (i.e.: What is Q2? Where is connected R09? etc). Also there are a lot of values hard to read, or unknow (most capacitors, some resistors, etc). Perhaps someone with a multitester and a board? I've used all @u22br 's photos. 
Iv'e used KiCAD, and some of the components are custom made, just because they don't exist in the library.
[DPS5015.zip](https://github.com/kanflo/opendps/files/4689215/DPS5015.zip)

kanflo, 2020-06-21:
I would say that unless you live in a territory where the Chinese government is in control and the DPS creators have "friends in high places" there is no chance you will have any trouble resulting form reverse engineering ;)

keyz182, 2021-02-14:
I also have the 2.3 board, and am having trouble. When I try to enable power, the screen will display 0V and 5.09A every time. 0V measured on the output at any time.

I've tried calibration, but when it gets to the `Calibration Point 1 of 2, 10% of Max` and the next `90%` step, I just get 1mV and the next stp fails (`ValueError: cannot convert float NaN to integer`).

I have the pre flash dumps [here](https://gist.github.com/keyz182/6d8fcba964ed10bcd7c7bb60b06e8424).

Kiotheka, 2022-11-19:
hello.
Ii have a contribution
IC U7 is a 24C02 memory. The RL resistor is the pull-up of the SDA pin

haarp, 2023-06-25:
Can anyone identify the output voltage sense line from the board? Would be useful for hacking in remote sense.

hieuktdt, 2024-06-06:
@kanflo, I have a problem when trying to change the font, because I'm currently using python3 so I can't build the gen_lookup.py file. I hope to get some help.

kanflo, 2024-06-06:
@hieuktdt Please do not reuse old threads for new issues. The main reason is that everyone in this thread now received an email that they most likely considered of little interest. Thanks for bringing it to my attention though, I have ported the font generator to Python 3. I will close this issue to prevent further derailing of the main discussion :wink:

closed 2024-06-06 by kanflo