kanflo/opendps #51: On/Off toggle on short circuit

f0086, 2018-09-28:
When I connect plus and minus, the background flashes white and it turned of. If I then got another short circuit (connecting plus and minus again), the device turned ON, and stays ON!

Maybe this is related to #42 and #36 ?

kanflo, 2018-10-01:
That would _really_ no be the intended behaviour. I will have a look at it, thanks.

kanflo, 2018-10-01:
Yes, #42 fixed it. I actually fried a '5005 in the process, should probably sign up at patreon.com...

Cabalist, 2018-10-01:
Please do!  You'd have my subscription.  :)

kanflo, 2018-10-01:
Thanks! I tried but the sign-up part of Patreon is down for maintenance :D

kanflo, 2018-10-01:
Gah. Encountered the issue when testing #49 that obviously does not carry the #42 fix. Probably time to call it a day, good night all.

closed 2018-10-01 by kanflo

Xenoamor, 2018-10-01:
+1 for patreon