kanflo/opendps #58: Cannot upgrade DPS5005

iondulgheru, 2018-11-16:
I have this DPS5005:
https://www.banggood.com/RD-DPS5005-50V-5A-Communication-Function-Constant-Voltage-Current-Step-down-Power-Supply-Module-p-1161704.html

I followed the instructions from the Readme in order to upgrade it but without success.
The errors I get are:

```root@ubuntu:/opt/kit/opendps# dpsctl/dpsctl.py -v -d /dev/ttyUSB0 -U opendps/opendps.bin
Traceback (most recent call last):
  File "dpsctl/dpsctl.py", line 617, in <module>
    main()
  File "dpsctl/dpsctl.py", line 612, in main
    handle_commands(args)
  File "dpsctl/dpsctl.py", line 385, in handle_commands
    run_upgrade(comms, args.firmware, args)
  File "dpsctl/dpsctl.py", line 451, in run_upgrade
    ret_dict = communicate(comms, create_upgrade_start(chunk_size, crc), args)
  File "dpsctl/dpsctl.py", line 349, in communicate
    if not comms.open():
  File "dpsctl/dpsctl.py", line 102, in open
    self._port_handle.open()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 286, in open
    self._update_dtr_state()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 110] Connection timed out
root@ubuntu:/opt/kit/opendps# dpsctl/dpsctl.py -v -d /dev/ttyUSB0 -U opendps/opendps.bin
Communicating with /dev/ttyUSB0
TX  9 bytes [7e 09 04 00 d2 26 55 6c 7f]
Error: timeout talking to device /dev/ttyUSB0.```

I checked and the TTL to USB device name is ttyUSB0.
I also ran these commands with success:
```
make -C libopencm3
make -C opendps bin
pip install pyserial
pip install pycrc
```
I tried on Ubuntu 18.04 and also on Fedora 28.
Any ideas?
Thank you

kanflo, 2018-11-16:
From your instructions it's not clear that you flashed the dpsboot.bin and opendps.bin binaries, did you?

iondulgheru, 2018-11-18:
@kanflo, I guess I didn't.
Can thi be done over UART? Because I have the model with 1.25mm spaced JTAG pins.
From the documentation I understand it can be done only with JTAG

kanflo, 2018-11-20:
That is true, you can only flash OpenDPS and its bootloader via JTAG.

kanflo, 2018-11-21:
Yes, you need to solder JTAG for the initial upgrade. When completed you can upgrade over the UART or via JTAG if you choose to keep the soldered pins. On the newer DPS:es soldering is a pain which is why I created the [DPS bottle](https://johan.kanflo.com/easing-the-pain-of-swd-on-modern-dpses/).

iondulgheru, 2018-11-21:
Thank you very much @kanflo !

ricktendo, 2018-11-21:
I just used some rainbow colored ribbon cable pre-tinned the wires so they are stiff and can fit through the holes, then I crimped some DuPont female pin headers on the other end

closed 2018-12-14 by kanflo