kanflo/opendps #59: Dual Use of "serial_handle_rx_char"

herostrat, 2018-11-28:
Hey guys,

I wanted to play a bit with SCPI a implementation, although I am at the very beginning and currently trying to understand this project structure.

My idea was to make a Makefile definition to use either SCPI or the currently used protocol.
From:
https://github.com/kanflo/opendps/blob/d906c1034fd8fb3a06a72e3fd8a8418de215692f/opendps/opendps.c#L717

I come to:
https://github.com/kanflo/opendps/blob/d906c1034fd8fb3a06a72e3fd8a8418de215692f/opendps/serialhandler.h#L28

And than there are two implementations:
 1. https://github.com/kanflo/opendps/blob/c9b7ecae9b0b22cf52c47e66a6768e45ee81838d/opendps/protocol_handler.c#L408
 2. https://github.com/kanflo/opendps/blob/4588761c30f39651088a6a2ab5b7727927a85408/opendps/command_handler.c#L76

As far as I understand it command_handler.c is used to start the execution of the received commands.
protocol_handler.c parses the received payloads and also executes the commands?

I guess I don't understand how two functions can have two same definitions and when which is executed.

Thanks for the help.

kanflo, 2018-11-29:
Cool! Your understanding is correct and the answer is in the [makefile](https://github.com/kanflo/opendps/blob/4588761c30f39651088a6a2ab5b7727927a85408/opendps/Makefile#L73):

```
ifeq ($(COMMANDLINE),1)
	CFLAGS +=-DCONFIG_COMMANDLINE
	OBJS += cli.o command_handler.o
else
	CFLAGS +=-DCONFIG_SERIAL_PROTOCOL
	OBJS += uframe.o protocol.o protocol_handler.o
endif
```

The ```command_handler``` implementation is best suitable for SCPI adaptions, the ```protocol_handler``` is used for the binary serial protocol used by ```dpsctl```. Building with ```make COMMANDLINE=1```will include the command handler rather than the protocol handler.

kanflo, 2018-12-14:
Closing as the dual use of ``serial_handle_rx_char`` was intended ;)

closed 2018-12-14 by kanflo