kanflo/opendps #60: DPS5005 Communication Module doesn't work at 115200 Baudrate

Xenoamor, 2018-12-15:
Looks like the isolated serial adapter that comes with the DPS5005 from this [Aliexpress link](https://www.aliexpress.com/item/RD-DPS5005-Communication-Function-Constant-Voltage-current-Step-down-Power-Supply-module-buck-Voltage-converter-LCD/1000004290003.html?spm=a2g0s.9042311.0.0.27424c4dgjoBBV) doesn't operate at 115200 baud. Changing the default baudrate of opendps and dpsboot to 9600 fixes this problem.

It appears to be a CH341 IC that communicates through 2x EL357N optocouplers. It is the optocouplers that are too slow to respond

I'm not sure if this is worth changing for the general users as I have no idea how many people actually use this supplied device. One possible solution would be to have the firmware default to 9600 and then have a command that can renegotiate a different baudrate

kanflo, 2018-12-15:
That would include the esp8266 bridge too but I am somewhat reluctant to this kind of change as it would probably cause confusion when things fail following a git pull and force everyone to upgrade everything on all devices.

kanflo, 2018-12-22:
@f0086 It does not respond at all? If so I need to try and reproduce that. By design, the bootloader shall not try to boot the opendps application if an upgrade failed or did not finish. However, you can force the bootloader to attempt an upgrade if you keep the SEL button pressed while powering on the DPS.

Xenoamor, 2018-12-24:
I don't think this is the same issue I was seeing. I couldn't communicate with the device at all
On a side-note I had an annoying issue with running Ubuntu in VMWare where the serial port would crash after one message. Running it outside of a virtual machine worked fine although I don't think this has anything to do with the python tools

Xenoamor, 2018-12-27:
I'll look into adding this after https://github.com/kanflo/opendps/issues/17 is resolved as I've not branched properly on my fork and will need to recreate it...

Xenoamor, 2019-01-04:
Scratch that I managed to create an upstream master branch that follows this one and I can now branch from that

PR #75 (2019-01-04): Allow changing of baudrate at compile time
This allows the changing of the serial baudrate used via the makefile e.g. `make BAUDRATE=9600`

It also allows a user defined baudrate to be used by dpsctl e.g. `python dpsctl.py -b 9600` or `python dpsctl.py --baudrate 9600`

I haven't tested the esp8266 stuff as I don't have the toolchain or the device but I believe this is the correct way to do it

This is to close #60
closed 2019-01-04 by kanflo