kanflo/opendps #65: Merge CV and CC modes

Xenoamor, 2019-01-01:
Most, if not all, benchtop power supplies operate by setting both a voltage and a current level and limiting whichever one is reached first. This is in effect operating seemlessly in both CV and CC modes

I sat down and watched [Dave's great video series on designing a PSU](https://www.youtube.com/watch?v=CIGjActDeoM&list=PLBF35875F73B5C9B5) and I believe we have a circuit similar to the one below (which can be found in the TL594 datasheet):
[image]

The way this works is it uses the feedback from the output to regulate the voltage. However the way the current limiting works is it uses the error amplifier as a comparator without feedback. This is on the contrary to the DPS3003 schematics that are present in this repo.

To test this I made [some minor changes to func_cv.c](https://github.com/Xenoamor/opendps/commit/73b334986b0626d68c8d136809f94299f4b724b5) so that both the voltage and current DACs can be controlled directly. This now works exactly as I would I expect it to.

In my opinion this is how the device should work. Although this does function it is still very important to calibrate the device and some kind of indicator would be necessary to indicate when the device is in CC mode. This could be done by turning the current value on the screen red using https://github.com/kanflo/opendps/pull/49 for example

It is worth noting that some sort of software feedback loop is still required however as at high current loads the voltages drift from it's target set point. This appears to be a steady state error and employing just the "I" (integral) part of a PID would correct this. As there is already a hardware feedback loop I don't believe employing the "P" (proportional) or "D" (derivative) sections is necessary or really a good idea as the two systems may fight each other and cause instability.

kanflo, 2019-01-01:
Thanks, I'll watch Dave's design videos (3.5 hours of presumably good stuff I see, amazing guy) and try to get my head around merging the CV and CC modes.

Xenoamor, 2019-01-01:
Part 1 shows all about how the current limiting works by using a comparator to turn off the output when the current exceeds the set value. [The code I linked to](https://github.com/Xenoamor/opendps/commit/73b334986b0626d68c8d136809f94299f4b724b5) does do what it should with them merged but it's worth watching that video as the penny didn't drop for me until I had

tzarc, 2019-01-02:
Feel free to take the PID control stuff from my branch -- it's working for CV but I've got no test loads available for anything to do with CC. I'd much prefer a more robust version, and as I've got no idea if my stuff breaks the CC stuff it'd be best if someone else could work out if it's broken.

Xenoamor, 2019-01-02:
Awesome thanks! I believe what you have for CV should work well. The only issue is I now need to be able to tell when the device is in CC mode. This is because in CC mode the device won't be able to raise the voltage anymore and the PID loop would run away with the voltage controlling DAC output.

Now the way this works in Dave's PSU is he has a digital IO pin connected to the output of the constant current limiting comparator. It's labelled as CLIM in the bottom right of this schematic:
[image]

Now I don't believe we have something like this on the DPS models (although I could be wrong). This means it's more tricky to determine which mode we are in but I've watched some youtube videos of the stock DPS5005 and I believe it changes from CV to CC on the display depending on which mode it is automatically so it can be done. It's just figuring out how they've done this

Xenoamor, 2019-01-02:
Okay I've [bodged it in](https://github.com/Xenoamor/opendps/commit/fd83fece85fa931d579103c41db3400fc0d2fa1d) to test it out and it now correctly displays either CC or CV depending on what it's doing. There's no actual feedback from the hardware loop that I can use that I'm aware of (I did check all the GPIO pins) so this works by determining which target setting is the one getting capped

Now this is fine for feedback to a person but it's not going to be accurate enough for a voltage PID

Xenoamor, 2019-01-05:
It's interesting to see that @jimmyw has reached the same conclusion with his commits [one](https://github.com/jimmyw/opendps/commit/a15d7dd058195bf393c0d7c13fa89c4457da7f63) and [two](https://github.com/jimmyw/opendps/commit/c139a538bca77ddbe1e599ebc57b85e8e014861e)

Xenoamor, 2019-01-06:
Okay this is quite easily possible but we'd have to remove the CC mode and perhaps rename the CV mode to main display or something. It would render the multiple screen functionality a bit redundant for the moment although I definitely see us adding in a settings screen at some point so we shouldn't remove it

What's your thoughts on this @kanflo? I'm happy to make a pull request for it but there will be quite a few changes as we'll be rename and removing a few things

jimmyw, 2019-01-06:
I think @Xenoamor is on to something here!

Did you check my layout that have all four (set voltage, set current, measured voltage, measured current) all in the same screen? Really think its the most simple ui possible, and wont hide anything. Adding a symbol that displays when the PSU switches from CV to CC mode would also be great, but if you se the set and measured voltage next to each one, its is quite obvious when it happens.

Also, a pid loop would be really great! @tzarc

Xenoamor, 2019-01-06:
You wouldn't happen to have a picture of it would you? I'm away from my device at the moment

I must admit I only got this device because I knew OpenDPS had a much more minimal display then the stock firmware. I'll make a mock up of what I'd personally like it to be when I'm back

tzarc, 2019-01-07:
Or, perhaps we pivot with the coloured fonts, and show limited outputs as orange or something instead?

Xenoamor, 2019-01-07:
I think our best option is to remove func_cv.c/h and func_cc.c/h and make a new pair of files called screen_main.c/h. This new screen will have the new combined mode in and would/should be easy to change and switch out for anyone who wants a custom display
@kanflo has already laid a great framework with his uui work so this should be easy to do

jimmyw, 2019-01-07:
I vote for keeping the screens option, but keeping the functionally the same.

So of some one wants the measure in watts, then can change to that screen, if some one wants in amps, thats good too. If someone wants in color, that could also be a screen.

The only reason for keeping CV mode, is if you like the PSU to shutdown when reaching the limit instead of limiting the amperage. I can se a usecase for that when you are working on equipment where underwoltage could damage the device.

jimmyw, 2019-01-07:
@Xenoamor I don't have any screenshot, but i can try take a photo when i get home after work.

tzarc, 2019-01-07:
There's also the invariable future screen for charging LiPo batteries and the like, too. _Someone_ will do that, surely ;)

jimmyw, 2019-01-07:
Oh, yes! A lipo charger, with a bar that shows percentage, and estimated time to full. :)

Xenoamor, 2019-01-07:
I not only intend to use it for charging batteries but also for emulating batteries, although this might be done by a python script.

We can certainly keep the CV screen and have a merged screen but it might be a bit difficult to tell which one you are currently on.. Maybe we could implement a way to switch the current or value to a hard limit which switched the device off if reached? Perhaps whilst setting the value of one you can long press something and it would change the colour to show that the state had changed? This way we could still have a single display but with the functionality of the CV/CC and merged modes

Xenoamor, 2019-01-07:
Yeah see this is the issue...
I can fairly easily add in a third screen for the combined mode but it won't be very clear if you're on CC, CV or the merged mode. I guess we could make a new logo for it?

tzarc, 2019-01-07:
This is @jimmyw's layout, for reference:
[image]

Xenoamor, 2019-01-07:
That case is really rather awesome. I think we could use the screen space a bit better though. I've noticed on the official firmware that the text actually touches the edges

jimmyw, 2019-01-07:
@tzarc Seems a little buggy :) Both the display and overshoot on the amps 😱

Xenoamor, 2019-01-09:
I think I'm going to add this in as a third screen so people can play around with it and see how they feel about it

PR #124 (2019-02-17): Merge cv and cc mode into a new screen
This adds a new screen called `func_merged.c/.h` which is a combination of the CC mode and the CV mode. This is to finally close #65 

In CV mode:
[image]

In CC mode:
[image]

To implement some of this I have added two new available functions to `ui_screen`. These are `activated(void)` and `deactivated(void)`. These are called when the screen is first swapped to and when it is about to be swapped away from. This is very handy for one time drawing events such as static text as used in `settings_calibration.c` and for clearing any graphic remnants such as  [... cut]
closed 2019-02-17 by kanflo