kanflo/opendps #68: DPS8005 Support

makefu, 2019-01-02:
Hey all,

I would love to have my dps8005 supported, i attach the on/off files for 5,10,15V attached to a 20V power source.

I would love to understand how the values correlate to the inputs in `dps-model.h`.

[5v-off.txt](https://github.com/kanflo/opendps/files/2722060/5v-off.txt)
[5v-on.txt](https://github.com/kanflo/opendps/files/2722061/5v-on.txt)
[10v-on.txt](https://github.com/kanflo/opendps/files/2722062/10v-on.txt)
[10v-off.txt](https://github.com/kanflo/opendps/files/2722063/10v-off.txt)
[15v-on.txt](https://github.com/kanflo/opendps/files/2722064/15v-on.txt)
[15v-off.txt](https://github.com/kanflo/opendps/files/2722065/15v-off.txt)

cheers!

Xenoamor, 2019-01-02:
Hmm.. that's very odd. The DAC values are the same on all of the xxv-on.txt files.
This is different to how the other devices work.

Are you sure you collected these with a different output voltage actually set? I think you have to "confirm" the input value by pressing set or something. Looking at the files I'd guess that they're all set for the same output voltage/current

I don't actually have a DPS8005 and I've no idea what the circuitry is but I'd be surprised if it were this drastically different

makefu, 2019-01-02:
Hey @Xenoamor , yeah i am quite sure that we confirmed the output voltage. We checked the source code and discovered that setting the register `DHR12R1` via openocd does not set the the output voltage ( as described in `dsp-model.h` ).
Additionally the static values for calculating the input voltage do not match the model  ( in `opendps/pwrctl.c` ).

Xenoamor, 2019-01-03:
Hmmm.. well that's very concerning then. Perhaps the DPS8005 does not have a DAC controlled output then. I guess alternatives are PWM that's then smoothed to an analogue voltage (although this makes no sense on a device with a DAC), or there's a seperate device that handles the voltage/current setting that is controlled via SPI/UART/I2C. You may have to have a look at the PCB and see if things are now drastically different

The static values you mention are going to be phased out soon by #17 where you'll be able to run a routine to determine the correct values

makefu, 2019-01-03:
@Xenoamor i really do think that the hardware revision is different, unfortunately. It contains new-style connectors and there are different chips on the board. When i get to it i can make some pictures, unfortunately for viewing the middle plate it seems i need to de-solder the 8 pins to the display, which is always a huge pain.

Xenoamor, 2019-01-03:
If you do take pictures please grab as high resolution ones as possible as like you said. You really only want to take that display off the once
There's an annoying hardware design fault that we discovered in #7 so I'd be interested to know if this one gets around it. Also any output filtering and reverse voltage protection would be a very welcome addition

makefu, 2019-01-06:
Hey @Xenoamor , i wrote a teardown of the DPS8005 in my personal blog: https://euer.krebsco.de/teardown-and-disassembly-of-the-dps8005.html . I really hope this helps with adding support for the DPS8005!

cheers

Xenoamor, 2019-01-06:
Oh. Well now this looks very very similar to the [DPS3003 schematic](https://github.com/kanflo/opendps/blob/master/hardware/reverse-engineering/dps3003/DPS3003.pdf).
It's all based around the TLC594 PWM controller. It also uses the same SGM8582 dual opamp. They seem to have changed the 3.3V regulator from a M5333B to a HT7133-1. And it looks like the constant current output load is the same LM321 aka A63A

The processor is the same. Can you see if R8 near the TLC594 and R37 near the SGM8582 both connect to the DAC outputs on the STM32?

Xenoamor, 2019-01-06:
That D0 LED on the back of the mainboard looks interesting. I wonder if that is some sort of output load

makefu, 2019-01-06:
@Xenoamor I am unable to test the connectivity right now but i will try to to this tomorrow. The "failure" was that i essentially got some rough voltage measurement working on the display by tweaking some of the values but i was unable to actually set the output value. I am unsure if it is related to the ADC and DCA values not being correct or because of some internals working differently on the DSP8005.

What i did was to test with the values for the DPS5005, however i was unable to change the output voltage by setting the register described in `dps-model.h`. These were my last values (before desoldering):
```
#elif defined(DPS8005)
 #define ADC_CHA_IOUT_GOLDEN_VALUE  (0x45)
 //AMPERE
 #define A_ADC_K (double)0.713
 #define A_ADC_C (double)0.51
 #define A_DAC_K (double)0.652
 #define A_DAC_C (double)0.611

  // VOLTAGE
 #define V_DAC_K (double)5.072
 #define V_DAC_C (double)10.85

// maybe does the right thing tm
 #define V_ADC_K (double)26.04
 #define V_ADC_C (double)-58.5
 #define VIN_ADC_K (double)26.000
 #define VIN_ADC_C (double)00.112
```
I used your fork already to set the VIN_ADC_K and C values.
I also have not changed the Ampere values because i was unsure how to do it. Now to think of it i guess it would be trivial by measuring the voltage difference and a know resistor (ohms law).

I tried to get the raw values from the controller, however flashing with `#define COMMANDLINE 1` resulted in build errors. After fixing them by removing the respective functionality the controller went unresponsive on the UART port.I also was unsure how to actually use the commandline mode afterwards anyway ... :)

That was my last progress anyway.

makefu, 2019-01-06:
@kanflo @Xenoamor i will try this as soon as i have soldered everything together :) I think i tried changing some registers but i was not seeing any changes, however i could be wrong here. When i initially dumped the registers from the original firmware i was not seeing any changes in `DAC_DHR12R1` but maybe i did something wrong and was not enabling the port correctly, however i am pretty sure that i pressed "enable" together with a friend.
Another nice addition would be a way to read raw ADC values via the serial interface instead of the command-line one.
How does the commandline mode is actually working? will it create a prompt on the serial console as well?

Xenoamor, 2019-01-06:
I do actually have something in my master branch that allows this which we called a "calibration report" This is part of the work for #17. It effectively gives the following:
```
python dpsctl.py -d COM7 -cr
Calibration Report:
        A_ADC_K = 1.71300005913
        A_ADC_C = -118.510002136
        A_DAC_K = 0.652000010014
        A_DAC_C = 288.610992432
        V_ADC_K = 13.1639995575
        V_ADC_C = -100.750999451
        V_DAC_K = 0.0719999969006
        V_DAC_C = 1.85000002384
        VIN_ADC_K = 16.7460002899
        VIN_ADC_C = 64.1119995117
        VIN_ADC = 913
        VOUT_ADC = 7
        IOUT_ADC = 0
        IOUT_DAC = 0
        VOUT_DAC = 0
```

makefu, 2019-01-06:
@Xenoamor will do once i put the dps8005 together again, thanks for your help!

Xenoamor, 2019-01-07:
Just a heads up this is now in the master branch. Good luck!

petermonna, 2019-09-25:
> constant current output load

@Xenoamor  can you explain more about the constant current output load with LM321 circuit? why we need it? thanks

Xenoamor, 2019-09-25:
A lot of power supplies oscillate (are unstable) if their load is less then a certain amount. This constant current sink provides this minimum load at all times and helps to ensure stability.

I'm more of a software guy so I don't fully understand the control loop and phase margin reasons for this but Dave on the EEVBlog discusses this [in his power supply series here](https://youtu.be/6Otr1I0OR18?t=1958). I'm not sure that is the correct timestamp, as I can't check the audio at work, but it is the LM334 he's drawn in blue on the whiteboard.

petermonna, 2019-09-25:
@Xenoamor  thanks so much for your help, very Very detailed and easy to understand, please help me for the last question , can you help me about the function of mosfet AF4407 conected to Vin?  i send picture about it in dsp3003 schematic
[image]

Xenoamor, 2019-09-25:
I think the schematic might be a bit wrong there. My best guess is it is a current inrush limiter but it's missing a capacitor

petermonna, 2019-09-25:
> I think the schematic might be a bit wrong there. My best guess is it is a current inrush limiter but it's missing a capacitor

@Xenoamor  oh yes thanks so much, you help me a lot, i will check again missing capacitor in pcb of dps5005

polihedron, 2019-10-31:
#186

yashm, 2019-12-12:
+1 for DPS8005 support!

Please let me know how I can be of service! I have 34 DPS8005 units that I can commit to testing.

Thanks all!

klenSA, 2021-09-18:
Hellow!
I would like to know if the DPS8005 has been able to start up so far?

lolyinseo, 2022-03-15:
> Hey @Xenoamor , i wrote a teardown of the DPS8005 in my personal blog: https://euer.krebsco.de/teardown-and-disassembly-of-the-dps8005.html . I really hope this helps with adding support for the DPS8005!
> 
> cheers

HI. Please share DPS8005 board photo. Blog is down (

makefu, 2022-03-15:
> HI. Please share DPS8005 board photo. Blog is down (

Hey, thanks for the heads-up, i fixed the blog :)

lolyinseo, 2022-03-20:
> Hey, thanks for the heads-up, i fixed the blog :)

Tnx man. Alternative firmware for ВЗЫ8005 will be here. You can ask the author in the comments

hseibel12, 2022-03-23:
I was also thinking about getting dps8005 model working under opendps. Following further investigations within this thread I put my point of interest onto the DAC-outputs and the schematics arround TL594.

I can definitely confirm that DAC1-output (Pin 14) ends up in an open wire! R01 has been removed and there seems to be no other connection.  (ref. to @lolyinseo blog). Also voltage on pin 14 is always zero. Otherwise DAC2 works and I could measure a linear relationship between V_SET and my DMM. The big question seems to be how original firmware sets the output voltage. From my opinion we need a reliable schematic around the TL594 to find out how voltage setting works.

Has anyone of you guys investigated the difference for this model?

Picture: Missing R01
[image]

lolyinseo, 2022-03-23:
> Has anyone of you guys investigated the difference for this model?

All you need is here https://profimaxblog.ru/%d0%bf%d1%80%d0%be%d0%b1%d0%bb%d0%b5%d0%bc%d1%8b-%d1%81-dps8005/

klenSA, 2022-03-23:
Hellow!
Since there is movement on this topic, 
I will try to disassemble 8005 and trace. 
At the moment, I already know that the board 
has three layers (?? it's strange, usually 2, 4...). 
I also suspect that in 8005 dac_ch1 is not used due 
to insufficient DAC resolution. instead, a 16-bit PWM (PA11) with 
a low-pass filter (R03-C33-R02-C32 + OPA follower) is used,
 which feeds the signal to 1IN of the TL564 driver.
As soon as I reliably identify the trace, I will unsubscribe to you with the results.

> I was also thinking about getting dps8005 model working under opendps. Following further investigations within this thread I put my point of interest onto the DAC-outputs and the schematics arround [...]

klenSA, 2022-04-14:
[image]
[image]
[image]
[image]
[image]

klenSA, 2022-04-17:
[image]

creatron, 2023-03-05:
> @Xenoamor thanks so much for your help, very Very detailed and easy to understand, please help me for the last question , can you help me about the function of mosfet AF4407 conected to Vin? i send picture about it in dsp3003 schematic [image]

AF4407 is n P channel FET.  The purpose of the FET is to switch the output off if the input voltage to falls below the value where the control circuitry stops operating.  If the main input for instance is switched off it will protect any possible glitzes on the output, that can destroy the unit connected to the PSU output.