kanflo/opendps #88: Need a debounce on buttons

tzarc, 2019-01-07:
Sometimes hitting power button doesn't register, and it flips itself back on or off if it happens too quickly.

Xenoamor, 2019-01-07:
I'm surprised I haven't seen this issue myself. Although thinking you've turned the device off and haven't can be dangerous
[I'm not seeing any debouncing](https://github.com/kanflo/opendps/blob/c4e3b4aa55767f4c97104fd76dcc50f7c188d2e4/opendps/hw.c#L728-L739) but this should be an easy addition

Xenoamor, 2019-01-07:
I can't actually reproduce this but it would be good belt and braces to implement it anyway

kanflo, 2019-01-07:
There ought to be debouncing. I take some shame in that there isn't ;)

tzarc, 2019-01-11:
The reason you've not been seeing it might actually be because the colour logic takes a bit of extra time to apply the masking -- will give it a try back with the unmodified white colours to see if it's related.

Tends to happen relatively frequently though -- I can sit there for 2 mins toggling and it'll happen 5 or 6 times.

Xenoamor, 2019-01-11:
Don't worry about reproducing it. It should definitely be in there either way

tzarc, 2019-01-11:
Oh I'm not worried. Just point out that there may be a reason I'm seeing it a bit more often than others.
I'd imagine the same problem occurs with the other buttons, I just don't muck about with them as often as I do with the power one.

Will have a play to see if I can confirm/deny, at any rate. More testing = more robust software, right? :)

kanflo, 2019-06-16:
Closed by #148

closed 2019-06-16 by kanflo