kanflo/opendps #95: Test if glyphs and icons can be converted to 4bpp mono

tzarc, 2019-01-11:
Check to see if we can relatively easily use 4bpp mono and do runtime conversion to the bgr565 format for each of the glyphs. A lot of flash space is currently being taken up by fonts, and converting to a 4bpp would be a ~75% size reduction.

More of a note for myself... unless someone else wants to tackle it.

Waiting for #93 and #94 to land before attempting it in earnest.

Xenoamor, 2019-01-11:
I think only using this on fonts and not the logos would be good. That way people can still use multi-coloured images but only single coloured fonts which is an acceptable compromise in my opinion

Waiting for #94 would be a good idea as this brings a lot of changes to how the glphy generator works and there would be a sea of conflicts otherwise

I was going to work on #80 later today but I don't see that having any direct overlap with this

tzarc, 2019-01-11:
Fair point. That said, I still think generating the logos like power button and thermometer etc. using FontAwesome would simplify a lot of things, though.

Xenoamor, 2019-01-11:
I'm not familiar with FontAwesome, what benefits would this have over the png/bmp/gif solution in #94?

tzarc, 2019-01-11:
Simplicity, really.

This was the "font" I'd used for the icons back for my early revs of the coloured fonts PR, for example:
https://github.com/tzarc/opendps-font/blob/master/icons.png

I just figure that they're "standardised" icons in a well-known open source set of glyphs, so rather than do things different we could leverage existing tried and true imagery.

Xenoamor, 2019-01-11:
I guess that might be simpler for actually getting the graphics icons but it does limit them to being block colours. I'm not sure it would work for the splashscreen either

kanflo, 2019-01-14:
It would be interesting to see the performance penalty of going to 4bpp

Xenoamor, 2019-01-17:
I'm starting to think this could be really useful as we could do with a full/partial alphabet for things like the settings page. We'd likely only need this in small or medium though

tzarc, 2019-01-18:
Honestly, I think we're going to require 1bpp for a full font -- for the number of glyphs required, even if it's scaled down to [a-zA-Z0-9+-] etc. 4bpp will still be a large amount of flash space used up.
The decode stage will take longer (should be similar to the colours, and is _probably_ why debounce is way more noticeable for me) -- but it might be unavoidable if we need full support.

I think the best bet would be to find a small bitmap font and export it for the full-text stuff. No anti-aliasing, just pure 1bpp data.

Had a busy week, no time to look at this yet, unfortunately.

tzarc, 2019-01-29:
Added a PR for 4bpp fonts -- flash size seems to drop by just over 13k. 
I'd imagine it's far more feasible now to look at an entire alphabet's worth of glyphs, but I think it's still going to be required that we do 1bpp or something similar. 2bpp could potentially give us a 256*uint64_t lookup table, so 2K flash for a 2bpp font, assuming we don't need anything more than 3 levels of grey.

Will prototype the existing fonts with this PR as a base, just for curiosity.

tzarc, 2019-01-29:
2bpp shows very little difference on-screen, but further reduces the flash size down by 14.5k.
Changed the PR to use packed 2bpp instead.

tzarc, 2019-01-30:
Thanks for the merge!

Given the significant flash space saving by just doing the fonts, I think there's less relevance to giving images the same treatment. We don't use a lot of them, and they're potentially not monochrome.

We can always revisit images in the future, if flash usage still needs to be addressed.

Closing, will open a new one for images if/when we need it.

closed 2019-01-30 by tzarc

kanflo, 2019-01-30:
As I see it the biggest advantage is that adding new icons will be a breeze. I probably spent an hour on the darn thermometer icon in photoshop ;)