kanflo/opendps #100: DPS5005 Brick

nmaas87, 2019-01-13:
Hello, I tried to update my newer version DPS5005 (the one with the smaller swd interface). I installed everything on a Raspberry Pi 3, got the connection working and halt and erased the memory like described.
However, the error persisted and even after shutting down the DPS and reconnecting, trying again, I got this error:

````
Open On-Chip Debugger
> reset halt
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
> flash erase_address unlock 0x08000000 0x10000
stm32x device protected
failed erasing sectors 0 to 63
>
````

I cannot erase the memory, but the main program is gone, so nothing works anymore.

````
sudo openocd -d1 -f sysfsgpio-raspberrypi.cfg -f target/stm32f1x.cfg
Open On-Chip Debugger 0.10.0+dev-00650-g31ea7037 (2019-01-13-12:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
SysfsGPIO nums: swclk = 25, swdio = 24
swd
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Error: stm32f1x.cpu -- clearing lockup after double fault
Polling target stm32f1x.cpu failed, trying to reexamine
Error: stm32x device protected
Error: failed erasing sectors 0 to 63

target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Error: stm32x device protected
Error: failed erasing sectors 0 to 63
````

I cannot program it with make -C opendps flash:
````
Open On-Chip Debugger 0.10.0+dev-00650-g31ea7037 (2019-01-13-12:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
SysfsGPIO nums: swclk = 25, swdio = 24
swd
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Error: stm32f1x.cpu -- clearing lockup after double fault
Polling target stm32f1x.cpu failed, trying to reexamine
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
** Programming Started **
auto erase enabled
Error: timeout waiting for algorithm, a target reset is recommended
Error: flash write failed at address 0x8000000
Error: error writing to flash at address 0x08000000 at offset 0x00000000
embedded:startup.tcl:480: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 539
at file "embedded:startup.tcl", line 480
Error: stm32f1x.cpu -- clearing lockup after double fault
Polling target stm32f1x.cpu failed, trying to reexamine
````

[... 770 lines cut ...]
>>>> exti
EXTI settings
IMR      : 0x0000033e  [0x40010400]
EMR      : 0x00000000  [0x40010404]
RTSR     : 0x00000300  [0x40010408]
FTSR     : 0x0000033e  [0x4001040c]
SWIER    : 0x00000000  [0x40010410]
PR       : 0x00000000  [0x40010414]

>>>> rcc
RCC settings
CR       : 0x03004683  [0x40021000]
CFGR     : 0x0410000a  [0x40021004]
CIR      : 0x00000000  [0x40021008]
APB2RSTR : 0x00000000  [0x4002100c]
APB1RSTR : 0x00000000  [0x40021010]
AHBENR   : 0x00000014  [0x40021014]
APB2ENR  : 0x0000422d  [0x40021018]
APB1ENR  : 0x20004006  [0x4002101c]
BDCR     : 0x00000000  [0x40021020]
CSR      : 0x0c000000  [0x40021024]
CFGR2    : 0x00000000  [0x4002102c]
````

Best

Xenoamor, 2019-01-13:
I'm not familiar with OpenOCD as I use JLink but can you [give this a try](https://community.particle.io/t/stm32x-device-protected-error-on-electron-with-openocd/23900/2)?

nmaas87, 2019-01-13:
Thank you, I got it flashed now.

I tried a lot:
https://community.arm.com/iot/embedded/b/embedded-blog/posts/unlocking-the-stm32f4-discovery-board-with-openocd
https://sourceforge.net/p/openocd/mailman/openocd-devel/thread/20141226001106204055.1d9b1832%40gpio.dk/
https://www.brobwind.com/archives/1139
Something of that must have gotten it working in the end...

However, now the calibration is very off:

My Inputvoltage (48V) is shown as 68V.
Also, if I set the DPS5005 to 5V and enable the output, it will drop to 4.79V. But what is a real problem, is that it actually outputs 3.4V, as measured.

I saw there should be some way to calibrate, but I cannot figure it out. I got the firmware now set to 9600 BAUD and installed, as it did not work otherwise with any serial interface.

Is there a short description on what to do to get these errors fixed? Thanks a lot!

Xenoamor, 2019-01-13:
Okay this is how to set my coefficients:

```
% python dpsctl\dpsctl.py -d DEVICE -b 9600 -c A_ADC_K=1.70325279236 A_ADC_C=-115.116912842 A_DAC_K=0.662139832973
A_ADC_K: ok
A_ADC_C: ok
A_DAC_K: ok

% python dpsctl\dpsctl.py -d DEVICE -b 9600 -c A_DAC_C=238.729721069 V_ADC_K=13.2220907211 V_ADC_C=-84.8741073608
A_DAC_C: ok
V_ADC_K: ok
V_ADC_C: ok

% python dpsctl\dpsctl.py -d DEVICE -b 9600 -c V_DAC_K=0.0756310075521 V_DAC_C=0.419114351273 VIN_ADC_K=16.8350524902 VIN_ADC_C=59.6804122925
V_DAC_K: ok
V_DAC_C: ok
VIN_ADC_K: ok
VIN_ADC_C: ok
```

They need to be split like this due to a bug I've just found #101 

If this really messes up your device you can reset them with:
`% python dpsctl.py --calibration_reset`

nmaas87, 2019-01-14:
Thank you very much for your help :)!
Could you point me on how to calculate the correct coefficients?
At the moment it would be most important for me to get the output voltage correctly (so that a) the DPS displays the correct output voltage and does also output the correct one ;))
Thank you very much :)!

Xenoamor, 2019-01-14:
Hi @nmaas87. I'll make implementing this my top priority
The voltage calibration work is fairly easy to put in so I don't imagine this will take long and I'll have something by the end of the day

Xenoamor, 2019-01-14:
@nmaas87 I've wildly underestimated the amount of time this is going to take me. If you'd like to get your hands dirty I did make [a calibration routine over here](https://github.com/Xenoamor/opendps/blob/adbd69af11f66b9347dc27b3d2e50535e2ae5bfc/dpsctl/dpsctl.py#L551) and you'd likely be able to figure out from that how the constants are calculated if you did want to work them out manually

nmaas87, 2019-01-15:
@Xenoamor Thank you very much for your hint, I was able to do the Vin and Vout calibration :)! It worked quite well, so here is my little write up for people who might follow:

````
info:
all datapoints for the vin_adc etc have been read by using the dpsctl program which is included in this github repo with following command:

./dpsctl.py -d /dev/ttyUSB0 -b 9600 -cr
(-d tells which USB Serial Device, -b with 9600 baud, I baked the firmware for my DPS5005 with 9600 baud instead of the default 115200 to prevent errors - and -cr reads out calibration data)

vinput calibration:

v1 lower power (here 6V)
v2 higher power (here 30V)

v1 = 6000 mV
v1_vin_adc = 479

v2 = 30000 mV
v2_vin_adc = 2506

#Math out the calibration constants
k_adc = (v1-v2)/(data1['vin_adc']-data2['vin_adc'])
6000 - 30000 / (479-2506)
-24000 / -2027
k_adc = 11,840157868771583621114948199309

c_adc = v1-k_adc*data1['vin_adc']
6000 - 11,840157868771583621114948199309 * 479
c_adc = 328,56438085841144548593981253099 

program:
dpsctl.py -b 9600 -d /dev/ttyUSB0 -c VIN_ADC_K=11.8401578687 VIN_ADC_C=328.564380858

voutput calibration

vin = 10V

Calibration Point 1 of 2, 10% of Max Voltage:
1V, 1A
measured:
c1=676 mV
c1_vout_dac=74
c1_vout_adc=82

Calibration Point 2 of 2, 90% of Max
9V, 1A
measured:
c2=6090 mV
c2_vout_dac=650
c2_vout_adc=656

calculate:
k_dac = (c1_data['vout_dac']-c2_data['vout_dac'])/(c1-c2)
k_dac = 74-650/(676-6090)
-576/(-5414)
k_dac = 0,10639083856667898042113040265977
0.1063908385

c_dac = c1_data['vout_dac']-k_dac*c1
74-0,10639083856667898042113040265977*676
c_dac = 2,0797931289250092353158478019955
2.0797931289

k_adc = (c1-c2)/(c1_data['vout_adc']-c2_data['vout_adc'])
(676-6090)/(82-656)
(-5414)/(-574)
k_adc = 9,4320557491289198606271777003484
9.4320557491

c_adc = c1-k_adc*c1_data['vout_adc']
676- 9,4320557491289198606271777003484 * 82
c_adc = -97,428571428571428571428571428569
-97.4285714285

program:
./dpsctl.py -d /dev/ttyUSB0 -b 9600 -c V_ADC_K=9.4320557491 V_ADC_C=-97.4285714285
./dpsctl.py -d /dev/ttyUSB0 -b 9600 -c V_DAC_K=0.1063908385 V_DAC_C=2.0797931289
````

I still would need to do the current calibration, but did not got time so far :/

Xenoamor, 2019-01-15:
Thanks for the in-depth write-up @nmaas87! It's interesting to see that your calibration settings are so far from mine. There must be quite a few variants of the hardware floating about

nmaas87, 2019-01-16:
Seems to be like that :). I will monitor #17 and will try auto calibration - especially for current in CV mode as soon as it becomes available :). Thanks a lot for your help, I will close this issue.

closed 2019-01-16 by nmaas87

Xenoamor, 2019-02-12:
@nmaas87 #120 and #122 should be of interest to you