kanflo/opendps #101: Issuing a lot of calibration coefficients at once fails

Xenoamor, 2019-01-13:
If you try to issue too many calibration coefficients in one go the device fails to respond. I assume this is due to some max string length issue. This could either be fixed by the python tool splitting it up or the OpenDPS firmware permitting bigger strings

```
% python dpsctl\dpsctl.py -d COM19 -b 9600 -c A_ADC_K=1.70325279236 A_ADC_C=-115.116912842 A_DAC_K=0.662139832973 A_DAC_C=238.729721069 V_ADC_K=13.2220907211 V_ADC_C=-84.8741073608 V_DAC_K=0.0756310075521 V_DAC_C=0.419114351273 VIN_ADC_K=16.8350524902 VIN_ADC_C=59.6804122925
Error: timeout talking to device COM19.
```

geekbozu, 2019-01-14:
I probably should have mentioned this, My apologies.
The issue comes from a string limitation on the receiving code size. It parses out the string DPS side and only has so much memory allocated to the string buffer. The quick solution to this is to separate it out inside the python code. Handling this should probably be done in the communications portion where is processes the supplied arguments so that changing modes and setting values is also handled appropriately.

Xenoamor, 2019-01-14:
Thanks for the information @geekbozu this makes sense. I'm going to be doing some work on `dpsctl.py` for #100 so I'll sort this out afterwards as it's fairly minor