kanflo/opendps #103: Extend dpsctl.py -c to allow setting of DAC values

Xenoamor, 2019-01-14:
Extend `dpsctl.py -c` to allow setting the DACs like so `dpsctl.py -c VOUT_DAC=10 `

This would be a useful addition to working towards completing #17

Xenoamor, 2019-01-14:
This could actually be done by creating a new UUI screen called `func_calibration.c`. This might be a better way of doing it as then we could use `dpsctl.py p VOUT_DAC=10`. This is also good as we can stop the screen "updating" the DAC values

Xenoamor, 2019-01-14:
Yes this would be a neat idea. I was thinking ideally it would be some _hidden_ screen accessed either by a special button combination or only from the python tools. 

I have started on this currently, it doesn't look like much but the top number is the vout_dac and the bottom is the iout_dac (both at their maximum 4095 values). Now I'm hitting a lot of hard limitations with the `uui_number.c` implementation but I'm learning a lot about what these are so I can improve them in #98 and I will continue to document these in that request.
[image]

Can you explain a bit about how the `main_ui` and `func_ui` @kanflo? My main thinking behind using something under `func_ui` would be so that we can use the following from `dpsctl.py`:
```
% python dpsctl.py -f calibration // to swap to the calibration screen
% python dpsctl.py -p V_DAC=4095 // to change the parameters on that screen
```

Currently this works fine but I guess we'd have to put more commands in if we were to make a new `uui_t`?

Xenoamor, 2019-01-15:
This makes perfect sense to me. I don't think we should use the screen set and get parameter functions that we currently have though as it doesn't really make sense from the python tools to have to change to the settings screen to a) list the available settings and b) change those settings. I think we should make two new python functions:
%dpsctl.py -S // Lists the available settings
%dpsctl.py -s setting=<value> // Sets the given setting

This gives us a good platform for when we inevitably need adjustable settings such as backlights, current-surge protection/allowance etc...

We would also need a function to switch to the calibration screen so that the DAC outputs don't get continuously updated by `cv_tick` or `cc_tick`

kanflo, 2019-01-16:
Ideally, when `func_ui` is hidden, it's screen tick functions should not get called, but rather those of the upcoming `settings_ui`.

kanflo, 2019-01-17:
Looks nice! I got an email tonight that go me thinking. The question was raised if it was possible to deduct the calibration data from the original firmware using ocd-client.py. I am thinking it should (after all that's what I did in the beginning of this project :). Should we perhaps work on calculating the calibration data from the ocd-client.py dump files and feeding it to the device? It would simplify calibration for those lacking mulitmeters.

Xenoamor, 2019-01-18:
That's a really nice idea. Yes we absolutely could do it. We'd have to have either the user read off the information or use the official firmware serial communication

Xenoamor, 2019-01-26:
Hmm, I'm having a hard time figuring out what commands to have for this from the dpsctl.py tool. Any ideas?

kanflo, 2019-01-26:
How about a switch to jump to and from a "settings mode" and when there, the settings mode has "functions" similar to CV and CC?

Xenoamor, 2019-01-26:
And share the current -F -f -P -p commands? That would be an easy change for sure

kanflo, 2019-01-26:
Yes, sharing -F and friends will fit quite well I think.

tzarc, 2019-01-30:
Just for future reference I guess, I generated an ascii-capable font using the 2bpp fonts branch:
https://github.com/tzarc/opendps/blob/2bpp-fonts%2Basciifont/opendps/font-settings.c

Used a bitmap font for it, brings it down to about 2k flash space for the entire font including metrics.

Xenoamor, 2019-01-30:
This is awesome. I guess we should hold off on this until #116 is merged in as currently this screen is unreadable. 2k is really a small price to pay as we can use this font for all sorts of things including this

PR #114 (2019-02-07): Calibration screen
This should hopefully close #103

The settings ui currently comprises of a single screen called calibration which has a cross-hair for an icon. The layout is as follows:
```
V_DAC    V_OUT_ADC
A_DAC    A_OUT_ADC
         V_IN_ADC
```
This will be clearer once we have a full alphabet available to use.

You can switch to the screen by either pressing both M1 and M2 or by issuing the command: `python dpsctl.py --screen settings` you may switch back to the main screen with `python dpsctl.py --screen main`

All the `-F -f -P -p -q` options work from this screen

This will also be [... cut]
closed 2019-02-07 by kanflo