kanflo/opendps #104: DPS5005 v2.3 Voltage Offset

serialx, 2019-01-15:
First, awesome project! Thank you! :)

I've upgraded my DPS5005 to OpenDPS according to the blog post. I'm seeing some voltage difference between what I set and what actually delivers.

DPS5005 Version (in the silkscreen): V2.3
CV Voltage set: 5.00 V
DPS5005 display shows: 4.73 V
My multimeter shows: 4.673 V

What could be the problem?

Dump of peripheral settings before reflash:

[4V-off.txt](https://github.com/kanflo/opendps/files/2760044/4V-off.txt)
[4V-on.txt](https://github.com/kanflo/opendps/files/2760045/4V-on.txt)
[5V-off.txt](https://github.com/kanflo/opendps/files/2760046/5V-off.txt)
[5V-on.txt](https://github.com/kanflo/opendps/files/2760047/5V-on.txt)

Xenoamor, 2019-01-15:
We're currently working on this for #17

kanflo, 2019-01-15:
First, thank _you_ :)

I will mark this as a duplicate of #17.

closed 2019-01-15 by kanflo