kanflo/opendps #109: Extend ocd-client.py to fetch factory programmed calibration coefficients

Xenoamor, 2019-01-18:
As discussed in #103 a method of fetching/retrieving the calibration coefficients from the original DPS firmware would be very useful

kanflo, 2019-01-18:
[Here](https://docs.google.com/spreadsheets/d/1AhGsU_gvZjqZyr2ZYrnkz6BeUqMquzh9UNYoTqy_Zp4/edit?usp=sharing) is the google sheet I used when reverse engineering my very first DPS5005. Dialling in a few voltages and reading the DAC setting will give us the voltage coefficients but of course one will need a few resistors to get the current coefficients.

Xenoamor, 2019-01-18:
Could you not just short the output terminals and current limit it?

kanflo, 2019-01-20:
That could work for lower currents. I think I have a stock DPS somewhere, I'll try it.

DavidMez, 2019-01-20:
I have a stock DPS5015 and an oscilloscope. what do you need?

kanflo, 2019-02-04:
@DavidMez Thanks, I should try shortening the terminals in CC mode on my own DPS first since I am not sure about the survivability ;)

Xenoamor, 2019-02-05:
In _theory_ it should be fine. I'm happy to try it when I get home after work today for you, I have one with stock firmware one

kanflo, 2019-02-17:
Update: shortening terminals does not seem to elect new popes :)