kanflo/opendps #111: Overvoltage limit in CV?

nmaas87, 2019-01-20:
Good evening,

I did successfully calibrate my DPS5005 with a 30V powersource and precision multimeter. At home, I plugged in the correct powersource, a 48V plugpack / powerbrick. The DPS did work kind of sluggish, did not react to inputs and turned its screen white. Looks like some kind of Overvoltage protection. Using it with a 20V powerbrick works. I did actually use the 48V always with my DPS for more than one year, never had an issue. I think the parameters for the calibration have been not good enough to compensate for the 48V - and that maybe it thinks it got over 50V and triggers some kind of overvoltage limit on Vin? Is such a limit implemented? Could that be the case? Any way to calibrate my DPS with the bigger plugback? (I need about  36V+ for certain applications, hence I need the big input...)

Thanks :)

Xenoamor, 2019-01-20:
There is no overvoltage limit for the input voltage.
There is however a current limit that when reached will flash the screen white and disable the output. This is the A setting on the CV screen. If your calibration coeffecients are off for your current this could cause issues that trigger this

Also see issue #6 which also plays a factor in this

nmaas87, 2019-01-20:
Ok I was dumb, sorry! I just saw I still had the USB Dongle attached to the Unit, without the USB Dongle attached to a PC, just floating. It probably pulled the TTL lines low and glitched the Unit... As soon as I deattached it, it worked. Sorry. I came to the conclusion as soon as I realised it did not flash the WIFI Icon upon boot anymore and just hung - without reacting to any input - just then turning to an all white screen - or glitching to the white screen as soon as I hit the set button (not played without output enable, as this did not work at all...)

Leason learned: Only attach serial ttl interface if you want to use it >.<

Can be closed

Xenoamor, 2019-01-20:
That's odd... which USB dongle is this?

closed 2019-01-20 by kanflo