kanflo/opendps #119: Is there any way to make power-state persist across resets?

fake-name, 2019-02-10:
Basically, I have a use-case for some DPS units in a context where they'd be set to a voltage, and then just left as-is, with some occasional tuning. 

Basically, I need the output to resume the last state it was in after a power cycle. If the output was on, it should be on when reset. If it was off, it should be off after reset. 

I must admit I kind of assumed this was already possible, but either I can't figure out how to set it, or it's not currently an option. Actually, I can't figure out how to access any sort of settings. Is there a document anywhere that describes what key-combinations do what? *Can* you even configure the DPS without a serial connection?

Xenoamor, 2019-02-10:
You absolutely can configure the DPS without a serial connection. In most cases this is how it is used however other then voltage/current there are no other settings options (ignoring the WIP #103 #114)

It's unlikely we have considered remembering the devices output state. We do preserve the output voltage/current settings between power cycles but not if the output is on or not. This is the same as the stock firmware behaviour. That's not to say you couldn't modify the firmware however to include this however

kanflo, 2019-02-10:
As @Xenoamor says this is not a use case that has been considered. It would only require a few changes and an additional build flag for enabling the mode. Pull requests are always welcome and you seem to have the proficiency to pull this off ;)

Xenoamor, 2019-02-10:
I'd say this would be a nice candidate for a new settings screen but I think it might be a bit dangerous in case someone turns it on by accident

fake-name, 2019-02-10:
> This is the same as the stock firmware behaviour. 

The stock firmware (at least the one I had) could be configured to load a preset on startup that included the output state (which is what I basically needed).

I'd be interested in helping, but only if I can switch to C++. The source for how the GUI works right now is damn near impenetrable.

------

Other stuff that'd be nice:
 - Soft limits on max voltage/current. Basically, some way to limit the inputs so you can't set the values beyond a specific value. In my case, I have a 48V input, and I'm controlling a output that takes 0-24V, 

I wound up doing some really horrible hacking to trick my unit into doing what I wanted. It's not generic in any way, but it does work.

Xenoamor, 2019-02-12:
C++ would be better suited for the GUI for sure. Currently it's effectively a C implementation of classes that is used
I'm not sure if using a small amount of C++ is a good idea though, honestly I have no idea. Do you know any example projects that use C++ for embedded? I've been personally considering the switch for ages (not for OpenDPS per se)

tzarc, 2019-02-12:
Every Arduino project, for example, is C++.... even if it doesn't look like it at face value. I've used it almost exclusively for embedded stuff I've worked on in the last decade. That said, I have this weird recollection that @kanflo would prefer to keep it written in C (I had a quick look and can't remember where I saw it, so I could be imagining it).

Anyways, as far as the issue is concerned -- "configurable but not default on" would be the way I'd go -- I've already fried circuits because the DPS decided to run in CC instead of CV at one point. Safety by default should probably override these sorts of things.

kanflo, 2019-02-12:
C++ offers a lot of attractive abstraction and encapsulation that is cumbersome to implement in C. I haven't coded C++ since the late 90s which is why I went for C in this project, and would like to keep it that way. A full rewrite in C++ could be nice but mixing in C++ would just look odd. I cannot recall speaking out on this topic though so your "weird recollection" @tzarc is more like mind reading. Which _is_ weird :D

kanflo, 2019-06-16:
Head up, I will implement this mode in the coming weeks. I have a TFT screen with a broken 14V power adapter and will use a DPS as a power source. Obviously I do not want any other modes than a compile time constant voltage and a compile time current limit on that particular DPS.

Anyone with a suggestion of a name for this mode?