kanflo/opendps #130: Improve calibration gradient detection algorithm

Xenoamor, 2019-02-19:
The current gradient detection algorithm does not work correctly in a noisy environment:
[image]

This should be improved to prevent incorrect calibration

JackNewman12, 2019-09-04:
Potentially [kneed](https://github.com/arvkevi/kneed) could be used to find the knee/elbow for each calibration plot, not sure how robust it is. I guess it also depends if you want to add another library just for this feature.

Additionally,I found that my DPS5005 would display 15mA open circuit (after calibration). Connecting a multimeter I found that it was always reporting 15mA above the real power. 
As part of making the calibration more robust it might be better to measure the open circuit current ADC and use the max value (say over 1 second?) as the zero crossing point.

Xenoamor, 2019-09-04:
I've never seen kneed before, it's likely a lot more robust then whatever I've implemented. If we don't want people to have to `pip install` it we can always just include the .py file directly (with credit of course)

Good thinking on the open circuit test. I think we can likely achieve this by just making sure that the `V_DAC` value is set to 0 though so we wouldn't have to remove the load.
https://github.com/kanflo/opendps/blob/08331f4c7225a9d6c30c4361cdcc5acf99ff1b74/dpsctl/dpsctl.py#L865

Currently this line can give a value that is non-zero even when `output_voltage` is 0 which is not ideal. We should add a special case to ensure `V_DAC == 0` is true. We can then be certain that this value is ~100% accurate and force our line of best fit to go through it