kanflo/opendps #131: Add calibration routine information to README

Xenoamor, 2019-02-19:


kanflo, 2019-02-20:
I have been thinking if the readme should be turned into a github wiki. Pros/cons anyone?

Xenoamor, 2019-02-26:
I'm not actually familiar with either. It might take me a while to get round to this as I'm trying my hand at designing my own power supply

pierow2k, 2019-07-06:
Moving the documentation into the Wiki would certainly make it easier for others to contribute and improve the documentation.  

With the instructions being _(primarily)_ located on the [blog](https://johan.kanflo.com/upgrading-your-dps5005/)(s), it takes a fair amount of jumping back and forth between the GitHub repo and the blog while reading through the nearly 200 [comments](https://johan.kanflo.com/upgrading-your-dps5005/#comments) that are on the bottom of the blog page.

kanflo, 2019-07-13:
You are absolutely right @pierow2k, I'll add this to my todo list for the summer.