kanflo/opendps #132: DPS5005 black screen when firmware is compiled with gcc 7.3.0

bhuism, 2019-02-20:
My DPS5005 will not start, so it seems, just a black screen, when the firmware is compiled with gcc 7.3.0. It does work with the awesome docker build script.

Xenoamor, 2019-02-20:
Hmm.. that's interesting
Perhaps this a good chance to get Travis running and have it add releases to the github page

kanflo, 2019-02-20:
Yes, we really need CI. I have been thinking of setting up a CI in my workshop with DPS hardware, STLink, every single version of gcc, ... and test building, flashing, upgrading and operations.

nmaas87, 2019-04-25:
@kanflo Drone.io could be a good choice here, they do free plans for OpenSource projects and integrate with Github https://cloud.drone.io/ :)