kanflo/opendps #133: Add compile time flag for disabling CV screen

Xenoamor, 2019-02-20:
This would be handy for those of us who only want the merged screen

kanflo, 2019-02-20:
Good idea. On a similar topic I have been thinking we could use some kind of `./configure` like step that would set the font, font colors and desired functions. With this config a `git pull && make` would build the latest and greatest according to one's configuration.

sercona, 2022-03-13:
I did some ugly hacks just to remove the CV/CC indicators from the screen since the use-case for my builds are for non-tech users and not having those telltales means less questions to have to answer ;)

I'm also thinking about making the default screen show the set-value and not the actual dac value.  I think when enable-mode is off, then its the set-value that is shown for voltage, but I prefer to always see that  has there been any thought to an option to show it that way?