kanflo/opendps #134: Error upgrading oder UART | AttributeError: 'bytes' object has no attribute 'encode'

gh7531, 2019-02-20:
Hello, I'm trying to upgrade my DPS to test the new CC/CV Merged screen. 
I can connect to the DSP over UART and enable/disable power so the connection is working. But if I try to Upgrade the Firmware I get the following error:

> ./dpsctl/dpsctl.py -d /dev/ttyUSB0 -U opendps/opendps.bin
> 
> Traceback (most recent call last):
>   File "./dpsctl/dpsctl.py", line 1135, in <module>
>     main()
>   File "./dpsctl/dpsctl.py", line 1129, in main
>     handle_commands(args)
>   File "./dpsctl/dpsctl.py", line 450, in handle_commands
>     run_upgrade(comms, args.firmware, args)
>   File "./dpsctl/dpsctl.py", line 559, in run_upgrade
>     if content.encode('hex')[6:8] != "20" and not args.force:
> AttributeError: 'bytes' object has no attribute 'encode'

It's probably something simple I'm missing but I can't figure out what it is

Xenoamor, 2019-02-20:
Are you using Python 3 out of interest?

gh7531, 2019-02-20:
Yes I did use Python3. It seems to work now using Python2.7. 
Thanks a lot.

Xenoamor, 2019-02-20:
Oh. Can you leave this open?
Ideally we'd like to have it be python 3 compatible

kanflo, 2019-02-20:
Yes, it really should work with Python 3.

kanflo, 2019-03-10:
Closed by #137

closed 2019-03-10 by kanflo