kanflo/opendps #142: dpsctl.py working only for some commands

iondulgheru, 2019-04-27:
Hi guys
I need your help to solve on issue with my new "OpenDPS"5005.
I can do only the following using dpsctl.py:
- ping
- version
- firmware upgrade
So the UART communication is working, but something is faulty here.

```
[root@localhost opendps]# python3 dpsctl/dpsctl.py -d /dev/ttyUSB0 --ping -v
Communicating with /dev/ttyUSB0
TX  5 bytes [7e 01 10 21 7f]
RX  6 bytes [7e 81 01 38 88 7f]

Got pong from device
[root@localhost opendps]# python dpsctl/dpsctl.py -d /dev/ttyUSB0 --version -v
Communicating with /dev/ttyUSB0
TX  5 bytes [7e 11 02 10 7f]
RX 18 bytes [7e 91 01 61 30 66 66 04 00 61 30 66 66 02 00 52 85 7f]

BootDPS GIT Hash: a0ff
OpenDPS GIT Hash: a0ff
[root@localhost opendps]# python3 dpsctl/dpsctl.py -d /dev/ttyUSB0 --status
[root@localhost opendps]# python3 dpsctl/dpsctl.py -d /dev/ttyUSB0 --power on
[root@localhost opendps]#
[root@localhost opendps]# cat /etc/redhat-release
Fedora release 29 (Twenty Nine)
[root@localhost opendps]# uname -a
Linux localhost.localdomain 5.0.9-200.fc29.x86_64 #1 SMP Mon Apr 22 00:55:30 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
[root@localhost opendps]# python --version
Python 2.7.15
[root@localhost opendps]# gcc --version
gcc (GCC) 8.3.1 20190223 (Red Hat 8.3.1-2)
```
I have built the firmware using actually the docker image and build_binaries.sh script.
Any ideas?

m-kozlowski, 2019-04-28:
Wrong syntax?
try `-q` instead of `--status` and `-o on` instead of `--power on`

iondulgheru, 2019-04-28:
Sorry, my bad, I followed this older blog post:
https://johan.kanflo.com/upgrading-your-dps5005/
Now that I am looking into the dpsctl.py documentation I see that the parameters have changed.
Thank you @m-kozlowski !

closed 2019-04-28 by iondulgheru

kanflo, 2019-04-29:
@iondulgheru that would be my bad :) All things still valid from the old blog posts should go into a getup wiki, when time permits.