kanflo/opendps #143: esp8266-proxy build: use option -std=c99 or -std=gnu99 to compile your code

iondulgheru, 2019-04-28:
Hi guys

I am trying to build esp8266-proxy, but I get the following error:
```
[root@localhost esp8266-proxy]# make
CC /opt/kit/opendps/esp8266-proxy/common.c
In file included from /opt/kit/opendps/esp8266-proxy/common.c:4:0:
/opt/kit/opendps/esp8266-proxy/../opendps/uframe.c: In function 'uframe_extract_payload':
/opt/kit/opendps/esp8266-proxy/../opendps/uframe.c:58:9: error: 'for' loop initial declarations are only allowed in C99 mode
         for (uint32_t r = 1; r < length-1; r++) {
         ^
/opt/kit/opendps/esp8266-proxy/../opendps/uframe.c:58:9: note: use option -std=c99 or -std=gnu99 to compile your code
make: *** [esp-open-rtos/common.mk:219: build/program//common.o] Error 1
```

I am using Fedora 29

```
root@localhost esp8266-proxy]# gcc --version
gcc (GCC) 8.3.1 20190223 (Red Hat 8.3.1-2)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

[root@localhost esp8266-proxy]# /opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc --version
xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-60-g37b07f6f) 4.8.5
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Any idea on what might be the issue an how I could solve it?

m-kozlowski, 2019-04-28:
Probably the least intrusive way:
`echo "CPPFLAGS = -std=gnu99" >> esp8266-proxy/esp-open-rtos/local.mk`

iondulgheru, 2019-04-29:
Not elegant but efficient :)
Thank you @m-kozlowski !

closed 2019-04-29 by iondulgheru