kanflo/opendps #147: Support for DPS3003

cortmen, 2019-06-03:
Hello, 
is there a chance that the DPS 3003 will be supported?

Unfortunately I have already flashed my new device 3003
with https://github.com/kanflo/opendps/issues/144
"flash dpsboot.bin at address 0x8000000 and opendps.bin at address 0x8001400"

The menu interface is ok, but the values ​​do not fit at all.
before flashed all values ​​are correct.

6V like output 3.7v
12V like output 7.2V

I have generated several 5,6,12V on / off dumps

Unfortunately I have opendps only with
make -C opendps MODEL = DPS3005 MAX_CURRENT = 3100
can generate.

Model 3003 has no support. Error - Message..hw.c  

I tried to connect via UART.
This only works until the --ping, answer is "pong" ..
but no more information --status etc.

Can it be that all registers, GPIO ports, values not configured for the DPS3003 and supported by opendps?

[5V-off.txt](https://github.com/kanflo/opendps/files/3247077/5V-off.txt)
[5V-on.txt](https://github.com/kanflo/opendps/files/3247078/5V-on.txt)
[6V-off.txt](https://github.com/kanflo/opendps/files/3247079/6V-off.txt)
[6V-on.txt](https://github.com/kanflo/opendps/files/3247080/6V-on.txt)
[12V-off.txt](https://github.com/kanflo/opendps/files/3247081/12V-off.txt)
[12V-on.txt](https://github.com/kanflo/opendps/files/3247082/12V-on.txt)

Many thanks for the support
Corty

kanflo, 2019-06-10:
Have you tried the dpsctl.py calibration routine? I would be surprised if '3003 and '3005 differed that much hardware wise.

cortmen, 2019-06-10:
Thanks for the replay,

I tried to connect via UART.
dpsctl.py -d /dev/ttyUSB0 --ping
….answer is pong

but !
dpsctl.py --status    
no informations return on the screen to see. only blank line

I have three cables on  tx / rx / gnd  to USB - UART 
Power on the DPS with 12v, Display Ready
I'm a bit surprised, why are 3.3v on tx pin?
Is that correct?
Horst

kanflo, 2019-06-10:
Try running dpsctl.py in verbose mode making it dump the UART traffic to see what's going on. As for 3.3V on TX, that's  [fully normal](https://en.wikipedia.org/wiki/Asynchronous_serial_communication) [wikipedia].

cortmen, 2019-06-10:
hi,

#./dpsctl.py -q  -v -V -d /dev/ttyUSB0 --status
Func       : cv (on)
  current  : 100
  voltage  : 5000
V_in       : 15.94 V
V_out      : 5.02 V
I_out      : 0.006 A
BootDPS GIT Hash: a0ff
OpenDPS GIT Hash: a0ff

Tomorrow I will be the calibration Routine run again.
I will report the result.
Horst

kanflo, 2019-06-10:
That looks quite spot on :)

cortmen, 2019-06-11:
when i excute the ./dpsctl.py -d /dev/ttyUSB0 -C
For calibration you will need:
        A multimeter
        A known load capable of handling the required power
        A thick wire for shorting the output of the DPS
        2 stable input voltages
Please ensure nothing is connected to the output of the DPS before starting calibration!

Would you like to proceed? (y/n): y

1  Remove everything from the output of the DPS5005
2. Power the DPS5005 from a bench power supply set to ~7V
3. Multimeter the input terminals and input the value read in mV
4. Turn up the bench power supply to ~15V
5. Multimeter the input terminals and input the value read in mV
6. Multimeter the output terminals and input the value read in mV
7. Multimeter the output terminals again and input the value read in mV
8 Enter the max output current of your DPS device (e.g. 5000mA for the DPS5005)
ok no Errors..

But what is that?

Enter your load resistor in ohms 
Connect your load resistor to the output of the DPS
Proceed if your load resistor can take the required wattage
The device will now iterate through 10 voltages and use this to calibrate the current ADC

Horst

fisherman-cp, 2019-06-14:
Moin,
just flashed BootDPS GIT Hash: a0ff, OpenDPS GIT Hash: a0ff-dirty to a DPS3003 device.
With 17V supply voltage setting to 5.0V gives me 3.059V Vout, set to 10V -> 6.14V Vout and 15V ->9.24V. 100mA on multimeter -> 150mA at device.
Calibration with 33ohm as load results in  
 Calibration Report:
        A_ADC_K = 1.75100004673
        A_ADC_C = -1.10099995136
        A_DAC_K = 0.652999997139
        A_DAC_C = 262.5
        V_ADC_K = 13.1309995651
        V_ADC_C = -111.900001526
        V_DAC_K = 0.076099999249
        V_DAC_C = 2.28570008278
        VIN_ADC_K = 16.7460002899
        VIN_ADC_C = 64.1119995117
        VIN_ADC = 1016
        VOUT_ADC = 14
        IOUT_ADC = 1
        IOUT_DAC = 0
        VOUT_DAC = 0
but every set voltage setting gives no output voltage.
After ./dpsctl.py --calibration_reset and some messurements and calculations
./dpsctl.py -c V_DAC_K=0.122940224958 V_ADC_K=8.128088730797  A_ADC_K=1.161831588384
gives an initial working setup for my device iv CV mode.
Hope this information will help you.
regards
Claus

Xenoamor, 2019-06-14:
This is almost certainly caused by #130 :(

cortmen, 2019-06-14:
@fisherman-cp 

thank you for your reply.
This information will help me well.
./dpsctl.py -v -d /dev/ttyUSB0 -c V_DAC_K=0.122940224958 V_ADC_K=8.128088730797 A_ADC_K=1.161831588384

My DPS3003 runs within the values, measured with the multimeter.
Horst

cortmen, 2019-06-14:
Calibration Report:
        A_ADC_K = 1.16183161736
        A_ADC_C = -1.10099995136
        A_DAC_K = 0.652999997139
        A_DAC_C = 262.5
        V_ADC_K = 8.12808895111
        V_ADC_C = -111.900001526
        V_DAC_K = 0.122940227389
        V_DAC_C = 2.28570008278
        VIN_ADC_K = 16.7460002899
        VIN_ADC_C = 64.1119995117
        VIN_ADC = 946
        VOUT_ADC = 422
        IOUT_ADC = 2
        IOUT_DAC = 2287
        VOUT_DAC = 415

kanflo, 2019-06-16:
@fisherman-cp  and @cortmen Do your '3003s run in an acceptable manner with these coefficients?

fisherman-cp, 2019-06-17:
Hi Johan,
did a bit more accurate calibration and with following settings the device works in an acceptable manner, tested with different loads from 3.3 to 33 ohms.
Calibration Report:
        A_ADC_K = 1.16183161736
        A_ADC_C = -1.10099995136
        A_DAC_K = 0.95069116354
        A_DAC_C = 262.5
        V_ADC_K = 8.12808895111
        V_ADC_C = -111.900001526
        V_DAC_K = 0.123375669122
        V_DAC_C = 2.28570008278
        VIN_ADC_K = 16.7460002899
        VIN_ADC_C = 64.1119995117
Rerun the calibration after calibration_reset several times allways results in unusable settings :-(
Regards
Claus

cortmen, 2019-06-17:
For me, all values are not accurate enough even after the x calibration. Especially the power measurement fluctuates too much.
I already got a new device 3003, I will get along with the original FW.

And please point out in the documentary that an imperfect recalibration must take place.

horst

adrianmihalko, 2020-03-22:
I am interested in 3003 support. Cant we just donate so @kanflo can buy one and test it?

kanflo, 2020-07-12:
@adrianmihalko while a most generous suggestion I lack the bandwidth right now meaning I cannot accept donations as it will only make donors disappointed.

martinstep, 2021-02-06:
I have found that DPS3003 calibration parameters were problematic. Some special cases of values, e.g., 
A_ADC_K = 1.0000
A_ADC_C = -1.0000
were causing a nasty overflow in
`uint16_t pwrctl_calc_ilimit_adc(uint16_t i_limit_ma)
{
    float value = (i_limit_ma - a_adc_c_coef) / a_adc_k_coef + 1;
    if (value <= 0)
        return 0;
    else
        return value + 0.5f; // Add 0.5 so it is correctly rounded when it is truncated
}`

When the current limit `i_limit_ma` is not needed it is set to `0xffff`. The conversion of ADC value to [mA] in function `pwrctl_calc_ilimit_adc` might return a value higher than `0xffff`, but the return value is casted into `uint16_t`.
The bug resulted in a very low el. current limit in CC or CL function mode, which was constantly triggering the overcurrent protection.

This bug is now solved in #226.